Koenkk/zigbee2mqtt #4726: zb won;t start Error: SRSP - SYS - ping after 6000ms)

yaniv95, 2020-10-20:
H,

I'm having trouble starting ZB  after reboot.
running on rsbpi 3b with CC1352R.
I thought this have to do with corrupted SD, so i have upgraded from 1.8.0 to 1.15.using the upgrade guide but it won;t solved it,
i getting this error

 zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Oct 20 13:09:58 zigbee2mqtt-pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 3.
Oct 20 13:09:58 zigbee2mqtt-pi systemd[1]: Stopped zigbee2mqtt.
Oct 20 13:09:58 zigbee2mqtt-pi systemd[1]: Started zigbee2mqtt.
Oct 20 13:09:59 zigbee2mqtt-pi npm[2841]: > zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
Oct 20 13:09:59 zigbee2mqtt-pi npm[2841]: > node index.js
Oct 20 13:10:00 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:info  2020-10-20 13:10:00: Logging to console and directory: 'data/log/2020-10-20.13-10-00' filename: log.txt
Oct 20 13:10:00 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:debug 2020-10-20 13:10:00: Removing old log directory 'data/log/2020-10-20.13-06-17'
Oct 20 13:10:01 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:debug 2020-10-20 13:10:01: Loaded state from file /opt/zigbee2mqtt/data/state.json
Oct 20 13:10:01 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:info  2020-10-20 13:10:01: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
Oct 20 13:10:01 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:info  2020-10-20 13:10:01: Starting zigbee-herdsman...
Oct 20 13:10:01 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:debug 2020-10-20 13:10:01: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6756},"serialPort":{"path":"/dev/ttyACM0","rtscts":true}}'
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:error 2020-10-20 13:10:21: Error while starting zigbee-herdsman
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:error 2020-10-20 13:10:21: Failed to start zigbee
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:error 2020-10-20 13:10:21: Exiting...
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]: Zigbee2MQTT:error 2020-10-20 13:10:21: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]:     at Generator.throw (<anonymous>)
Oct 20 13:10:21 zigbee2mqtt-pi npm[2841]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! code ELIFECYCLE
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! errno 1
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! Exit status 1
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR!
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR! A complete log of this run can be found in:
Oct 20 13:10:22 zigbee2mqtt-pi npm[2841]: npm ERR!     /home/pi/.npm/_logs/2020-10-20T10_10_22_048Z-debug.log
Oct 20 13:10:22 zigbee2mqtt-pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 13:10:22 zigbee2mqtt-pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 20 13:10:22 zigbee2mqtt-pi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Oct 20 13:10:22 zigbee2mqtt-pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.

thanks for helping
Yaniv

Koenkk, 2020-10-20:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

ilouga, 2020-10-20:
I am having similar issue with CC2652, getting this error **a couple of times** after restart of zigbee2mqtt, then the zigbee-herdsman starts sometimes (sometimes not at all). Went through the guide above 

https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start, still having the issue unfortunately. 
Seems though that sometimes, plugging in the stick and unplugging it helps (sometimes).

Zigbee2MQTT:info  2020-10-20 23:52:58: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-20.23-52-58' filename: log.txt
Zigbee2MQTT:info  2020-10-20 23:52:59: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-20 23:52:59: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-10-20 23:53:25: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-20 23:53:25: Failed to start zigbee
Zigbee2MQTT:error 2020-10-20 23:53:25: Exiting...
Zigbee2MQTT:error 2020-10-20 23:53:25: Error: SRSP - SYS - osalNvRead after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm
 ERR! code ELIFECYCLE
npm
 ERR! errno 1
npm
 ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-20T21_53_25_321Z-debug.log
2020-10-20T23:53:25: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-10-20T23:53:25: PM2 log: App [npm:0] starting in -fork mode-
2020-10-20T23:53:25: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js

After the start (couple of seconds later)  I am also getting the following error:
error 2020-10-20 23:32:16: Failed to call 'Bridge' 'onMQTTConnected' (Error: SRSP - ZDO - extNwkInfo after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

anyone having same issue as well? 

Running on Hassio, Pi4, ver. 1.15.0, zig-a-zig-ah! (CC2652 Stick)

yaniv95, 2020-10-21:
1. fixed by reflashing  CC1352 
2. recover from backup (v0.1.8 ) and upgrade to 0.1.15

thanks for the quick support
:)

closed 2020-10-21 by yaniv95

ilouga, 2020-10-21:
so its a firmware issue in case of CC1352? I am using CC2652, maybe there is also a firmware issue?

yaniv95, 2020-10-21:
it seems to work even better, I have discovered lost entites :)

yaniv95, 2020-10-21:
I ran CC1352P_2_20191106 and now flashed to CC1352P_2_20200925