Koenkk/zigbee2mqtt #4727: All devices go unavailable

mf-social, 2020-10-20:
## What happened
For the last few weeks, every few days I get a full dropout of all my devices. 

## What did you expect to happen
Devices remain available. 

## How to reproduce it (minimal and precise)
Unknown, it just happens, see log below. 

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version: latest I think but I can't double check as I'm not at home 

When this happens and I look in the log (I have both debug logs and herdsman logs) everything seems to be fine until suddenly I get 'failed to ping' errors on all of my devices, one after the other over the course of a minute or so, and then the log just stops. Zigbee2mqtt is still 'running' there's just no log output and it seems to have just 'given up'. 

If I unplug and replug the dongle, herdsman restarts and everything will work fine again for a few days, and then I'm back to this. 

Last entry in the log is

```
Zigbee2MQTT:debug 2020-10-20 16:10:37: Failed to ping 'lr_left_light'
2020-10-20T15:10:37.075Z zigbee-herdsman:controller:endpoint Read 0x0017880102fd3867/11 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-10-20T15:10:37.075Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0017880102fd3867:38601/11 (0,0,11)
```

And basically the entries before it are the same, but for my other devices one by one. Prior to that it's just normal updates that I see all day everyday. 

Any help appreciated.  I've been running zigbee2mqtt for quite a while now and it's been rock solid, but this is causing quite an inconvenience and I don't know where to start 🙂

mf-social, 2020-10-20:
If it helps, it seems to drop out between 3 and 7 days, like it's just dropped out today before I posted this issue, the last time was Saturday just gone, then the 11th, then the 4th before that. It happened a couple of times before that too but I don't know the dates.

mf-social, 2020-10-20:
Just to confirm, my other half who is at home just unplugged the dongle and reconnected it and everything is back up.

Im running the hassio addon so I suspect my options are limited, but does anyone know if there is a way to restart herdsman without restarting the addon? I'm curious to know whether it is herdsman restarting it that fixes it, or if it is the act of disconnecting the dongle. 

All suggestions welcome.

Koenkk, 2020-10-20:
Try disabling the availability feature. (remove `availability: XX` from your configuration).

mf-social, 2020-10-20:
I don't have `availability:`

I have `report: true` and `availability_timeout: 6`

Both are under the `advanced:` key. 

Should I just completely remove one/both?

Koenkk, 2020-10-20:
Sorry, I meant `availability_timeout`,  6 is a very low value and causes a lot of stress on the network. I recommend to first remove this and see if your network keeps running.

mf-social, 2020-10-20:
I'll give it a go and see what happens. I'll come back to you with results good or bad, although to be sure of good result I'll have to wait at least a week.

Appreciate the prompt response @Koenkk , cheers 👍

mf-social, 2020-10-30:
Hey @Koenkk,

Hopefully I'm not tempting fate here, but it's been 10 days since I removed availability_timeout and no crashes, so it looks like you were correct in determining the cause.

Before I close this issue though, can you recommend a good figure I could use for availability_timeout that would not cause this? Obviously without it I do not get any reports of unavailable items. 

I presume it's in seconds, so maybe 30000? That would be 5 minutes, so presumably a device going offline would be notified to homeassistant within 5 minutes?

5 minutes is perfectly acceptable for me, but I don't want to start randomly guessing at what a decent interval would be. 

I have 5 battery operated motion sensors, 5 battery operated window/door sensors, 4 battery operated buttons/switches, 1 'green' switch, 12 mains powered bulbs and a CC26X2R1 controller if this makes any difference to the 'optimum' figure.

Thanks in advance for your advice 🙂

Koenkk, 2020-10-30:
300 should be fine, but the only way to find out is to test :)

mf-social, 2020-10-30:
Haha, I was hoping you might have a bit of a formula in your head for how much stress is caused on the network 😂

OK, I will chuck 300 at it and see how it goes 😃

Once again, thanks for responding so promptly to this issue and diagnosing it 👍

closed 2020-10-30 by mf-social