Koenkk/zigbee2mqtt #4731: Need to power-cycle CC26X2R1 after restart of 1.15.0

maichai, 2020-10-20:
## What happened

before 1.15.0 zigbee2mqtt ran and ran without me having to power cycle the CC26X2R1. I would only have to power cycle it after a reboot of the host machine.

Since 1.15.0 I need to power cycle the CC26X2R1 once a day and then restart the container to make it work again. Note: all contianers are stopped and started by some cron-job at least once per day (this is intentional).

## What did you expect to happen

1.15.0 should not need power cycle of the controller after container restart.

## How to reproduce it (minimal and precise)

I think restarting the container is enough, I am not sure.

After w few weeks of doing the power cycle exercise every day I got sick of it and went back to 1.14.4. No such problems with that version.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0

Koenkk, 2020-10-21:
Could you post the herdsman debug log with both 1.14.4 without problems and 1.15.0 with problems?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

cody82, 2020-10-28:
Maybe I have the same issue:
I recently switched from a CC2531 to the zzh stick. I also upgraded Zigbee2Mqtt, so I dont know what caused the issue.

When booting up zigbee2mqtt gets started but it can not recognize the zzh(timeout). The USB port ttyUSB1 is available. Starting zigbee2mqtt manually also does not work. Reloading the USB serial drivers to redetect the zzh also does not solve this.

The only thing that works is pulling the zzh out of the USB port and putting it back in. After that it is recognized again.
To me it looks like the firmware on the stick crashes and does not answer anymore.

If that is a different issue I will open a new one after investigating further.

closed 2020-12-05 by github-actions[bot]