Koenkk/zigbee2mqtt #4733: Zigbee2MQTT fails to start, Adapter dies after installing NUT

Anon0ne, 2020-10-21:
## What happened
Installed the NUT package for UPS monitoring, now the adapter is shutting off and failing to start. I verified it's the problem by unplugging the UPS, and errors were gone.

These are errors from dmesg:

```
[32921.793455] usb 2-1.8: device not accepting address 25, error -110
[32921.793641] usb 2-1-port8: unable to enumerate USB device

```

## What did you expect to happen
Normal start-up?

## How to reproduce it (minimal and precise)
Install NUT, use the USB adapter for an APC  BackUPS and try to restart Zigbee2MQTT....

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 20190608

Is there a way to setup UDEV or something to get around this? Not sure why the UPS monitor is killing my zigbee; but the USB HDD on the same USB bus aren't having issues thus far.

kitopopo, 2020-10-21:
Hello, I have Nut installed and zigbe2mqtt starts correctly. Make sure you haven't plugged the zigbee2mqtt into another usb port. If you change usb position you should check the tty0 configuration in the configuration.yaml file. I changed the usb port for a long time and zigbee2mqtt did not start. Luck

Anon0ne, 2020-10-29:
I suppose NUT isn't the problem. The only other thing I have changed recently is the VM guest OS was upgraded to Ubuntu 20.04; 

```
[ 9530.812436] usb 3-1: new full-speed USB device number 2 using uhci_hcd
[ 9530.990823] usb 3-1: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[ 9530.990825] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9530.990827] usb 3-1: Product: TI CC2531 USB CDC
[ 9530.990828] usb 3-1: Manufacturer: Texas Instruments
[ 9530.990829] usb 3-1: SerialNumber: __0X00124B0001DE4D91
[ 9531.547384] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[ 9531.549788] usbcore: registered new interface driver cdc_acm
[ 9531.549791] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

```

```
Zigbee2MQTT:info  2020-10-29 16:29:45: Logging to console and directory: '/app/data/log/2020-10-29.16-29-45' filename: log.txt

Zigbee2MQTT:debug 2020-10-29 16:29:45: Removing old log directory '/app/data/log/2020-10-29.16-27-04'

Zigbee2MQTT:debug 2020-10-29 16:29:45: Loaded state from file /app/data/state.json

Zigbee2MQTT:info  2020-10-29 16:29:45: Starting Zigbee2MQTT version 1.15.0 (commit #1bccc5d)

Zigbee2MQTT:info  2020-10-29 16:29:45: Starting zigbee-herdsman...

Zigbee2MQTT:debug 2020-10-29 16:29:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'

Zigbee2MQTT:error 2020-10-29 16:29:46: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2020-10-29 16:29:46: Failed to start zigbee

Zigbee2MQTT:error 2020-10-29 16:29:46: Exiting...

Zigbee2MQTT:error 2020-10-29 16:29:46: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'

    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)

    at Generator.next (<anonymous>)

    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71

    at new Promise (<anonymous>)

    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)

    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)

    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)

    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12

npm ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`

npm ERR! Exit status 1

npm ERR! 

npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.

npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

npm ERR!     /root/.npm/_logs/2020-10-29T20_29_46_262Z-debug.log

```

closed 2020-12-06 by github-actions[bot]