Koenkk/zigbee2mqtt #4735: Data request failed with error: 'undefined' (241)

liquid-motion, 2020-10-21:
## What happened
All of a sudden my complete Zigbee network is failing.

## How to reproduce it (minimal and precise)
- Network scan gives this LQI-error for all my devices:
`Zigbee2MQTT:error 2020-10-21 11:29:52: Failed to execute LQI for 'Spot-midden'`
- When I try to switch a light off, I get this message:
`Zigbee2MQTT:error 2020-10-21 11:31:08: Publish 'set' 'state' to 'Spot-keuken' failed: 'Error: Command 0x000d6ffffed64eed/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'undefined' (241))'`

## Debug info
Zigbee2MQTT version: 1.15.0 (on Hassio)
Adapter hardware: CC2531
Adapter firmware version: zStack12 (20190223)

Koenkk, 2020-10-22:
Does this still occur with the latest firmware? (I recommend this one: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

ljan, 2020-10-23:
I get the same errors since about 2 days
CC2531 firmware is revision: CC2531_DEFAULT_20190608.zip

liquid-motion, 2020-10-23:
> Does this still occur with the latest firmware? (I recommend this one: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

I have no idea. It was working fine with the firmware I'm using... until a few days ago.

jti989, 2020-11-08:
I am observing the same behavior since today.
Discovered on the latest-dev docker image from yesterday, downgraded to 1.15.0 (which worked with my stick before), but no luck.
CC2531 firmware is revision: CC2531_DEFAULT_20190608.zip

I can help with logs if you want, just let me know.

jti989, 2020-12-09:
I have reflashed the firmware, still using CC2531_DEFAULT_20190608, now it is works again.
Not sure what happened.

Mocem1, 2020-12-28:
I had the same issue. For me unplugging the USB and restarting HA, Zigbee2mqtt and mosquito solved the problem.

closed 2021-02-05 by github-actions[bot]