Koenkk/zigbee2mqtt #4736: Slaesh zigbee stick

cndian, 2020-10-21:
Hello,

I just received the Slaesh zigbee stick. I currently have the cc2531 stick.

I know I have to change the pan_id but have two questions about it.

Do I change the ID before unplug the old stick or can I put the new stick in first and then change the ID?

And when I've done I know I need to repair my devices. Should I first delete everything in domoticz or does it still know.my friendly names and settings?

Many thanks.

flrahe, 2020-10-21:
Did you get any reply? I will have to do the same, except I'm using Homeassistant...

cndian, 2020-10-21:
Not yet

artist67, 2020-10-21:
I would recommend to stop z2m, then change the pan_id, swap the sticks an start z2m again.
Also check if the new stick is visible as ttyACM0 or ttyUSB0 and update the config file accordingly.
As I'm using Openhab, I cannot help further.

Koenkk, 2020-10-22:
- Stop zigbee2mqtt
- Unplug CC2531, plug CC2652RB
- Update serial port path in `configuration.yaml` and change pan_id
- Start Zigbee2MQTT and start repairing
  - If repairing fails make sure to completely power down the old network first (power off all routers)

flrahe, 2020-10-22:
Thank you for the reply. I had one open question:Where to change the pan_id to.
But there is a extended thread telling more details (i wished I found it earlier): [Easiest way to migrate from CC1352P-2 to zzh?](https://zigbee2mqtt.discourse.group/t/easiest-way-to-migrate-from-cc1352p-2-to-zzh/1304)
Just increase it by 1.

Koenkk, 2020-10-22:
The default one is `0x162a`, so you can use `0x163a`

cndian, 2020-10-22:
Thanks Koen, 

So I don't have to delete the devices?

Koenkk, 2020-10-23:
@cndian nope, zigbee2mqtt will automatically reset the database when it creates a network (which is done when the panid changes)

closed 2020-12-01 by github-actions[bot]