Koenkk/zigbee2mqtt #4739: Lag / error messages since upgrade to zzh! CC2652R

Rido, 2020-10-22:
## What happened
Since the upgrade to the `zzh! CC2652R` adapter, I'm having problems with lag & error messages in the logs.
I did a full network reset after the upgrade, so I reconnected all devices. 
If I turn all the lights on or off in the living room, for example, there are often some lights left behind with a reaction, and only 1-2 minutes later, they react.

I created a pastebin with the log output for the light "Overloop 2", this one just had problems when I wanted to turn off all the lights: https://pastebin.com/s2pMfpr7
It's not always the same light with these problems, it's random, sometimes just 1 and another time 2 or 3.

## What did you expect to happen
When I toggle the lights, I expect everything to respond immediately. Now there are sometimes delays of 1-2 minutes.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2652R
Adapter firmware version: CC26X2R1_20200925

Koenkk, 2020-10-22:
- Is your zzh connected through a USB extension cable?
- What device is Overloop 2?
- Can you provide complete zigbee herdsman debug log from Zigbee2MQTT start till errors?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Rido, 2020-10-22:
@Koenkk Thank you for the reply.

The zzh adapter is directly connected to the Intel NUC.

I restarted Zigbee2mqtt with the Zigbee herdsman debug log enabled: https://pastebin.com/d8U55zWY
This time the **Staande lamp** didn't turn off, the Philips Hue model is 9290012573A.
**Overloop 2** is the Philips Hue model 8718696598283.

Koenkk, 2020-10-23:
I'm not sure if it will solve it, but to rule out any interference from the NUC, connect the adapter through a usb extension cable and position it 20cm+ from the NUC. Interference is a common cause of issues like this.

Rido, 2020-10-26:
Thanks for the tip @Koenkk!
I connected the adapter through a USB extension cable and the network is more stable now.

closed 2020-10-26 by Rido