Koenkk/zigbee2mqtt #4741: Device does not pass interview (OpenWRT)

redspirit, 2020-10-22:
# Bug Report
## What happened
Device does not pass interview

## How to reproduce it (minimal and precise)
I started OpenWRT (Entware mipselsf-k3.4 rep) on my internet router (Keenetic Extra II). I installed the node (v12.18.3), npm (6.14.6) and installed z2m. When I try to connect a sensor, I get an error in the log. The device does not pass the interview. I made all recommendations from the documentation. I used different versions of z2m (master and dev).
The important thing is that I ran this stick on my home server with ubuntu and everything worked well there and the devices were connected (motion sensor and smoke sensor). I want to make a router-based setup.

Log: [https://pastebin.com/bRTUAaDp](url)
I see an error on the line: 439
`zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '14127' failed (TypeError: Cannot convert undefined or null to object), retry +5s`

I myself compiled the serialport binding library, but I don't know if it affected the operation of the device.

## Debug Info
Zigbee2MQTT version: 1.15.0-dev
Adapter hardware: CC2531
Adapter firmware version: zStack12 (20190608)

Koenkk, 2020-10-23:
Please provide the logging again after the following modification to `/opt/zigbee2mqtt/node_modules/zigbee-hersdman/dist/adapter/z-stack/adapter/zStackAdapter.js` change `debug(`Node descriptor request for '${networkAddress}' failed (${error}), retry`);` to:

```
debug(`Node descriptor request for '${networkAddress}' failed (${error.stack}), retry`);
```

redspirit, 2020-10-24:
I replaced the specified line (added .stack in log)
[https://pastebin.com/kr3MjnX9](https://pastebin.com/kr3MjnX9)

[image]

redspirit, 2020-10-24:
I compared the interview log with a router and with a home server. It can be seen that the data is the same.

Ubuntu server:
[image]

OpenWRT:
[image]

nodeDescRsp object is also identical

Koenkk, 2020-10-24:
In `zStackAdapter.js` can you change the following:

```js
    nodeDescriptorInternal(networkAddress) {
        return __awaiter(this, void 0, void 0, function* () {
            const response = this.znp.waitFor(Type.AREQ, Subsystem.ZDO, 'nodeDescRsp', { nwkaddr: networkAddress });
            console.log('response', response);
            const payload = { dstaddr: networkAddress, nwkaddrofinterest: networkAddress };
            console.log('payload', payload);
            console.log('znp', this.znp);
            yield this.znp.request(Subsystem.ZDO, 'nodeDescReq', payload, response.ID);
            const descriptor = yield response.start().promise;
            let type = 'Unknown';
            const logicalType = descriptor.payload.logicaltype_cmplxdescavai_userdescavai & 0x07;
            for (const [key, value] of Object.entries(Constants.ZDO.deviceLogicalType)) {
                if (value === logicalType) {
                    if (key === 'COORDINATOR')
                        type = 'Coordinator';
                    else if (key === 'ROUTER')
                        type = 'Router';
                    else if (key === 'ENDDEVICE')
                        type = 'EndDevice';
                    break;
                }
            }
            return { manufacturerCode: descriptor.payload.manufacturercode, type };
        });
    }
```

redspirit, 2020-10-24:
[https://pastebin.com/j6d7qP6W](https://pastebin.com/j6d7qP6W) line 352

For me, these logs did not provide useful information. I'm not sure if I can install node 14 on the router, but I'll try to think of something. If I can do it, then I will write about it.

closed 2020-12-02 by github-actions[bot]