Koenkk/zigbee2mqtt #4743: Ikea Tradfri GU10 missing channels after pairing

jti989, 2020-10-22:
## What happened
After pairing an Ikea Tradfri GU10 warmwhite (https://www.zigbee2mqtt.io/devices/LED1650R5.html), it shows as connected and supported, but there is only the update channels, not the OnOff and Brightness Channels

## What did you expect to happen
See the State and Brightness Channels like I do on my Hue Bulbs and my Osram Smart+ Plugs.

## How to reproduce it (minimal and precise)
1. Pair Tradfri GU10.
2. Optional - wait for hours

## Debug info
Zigbee2MQTT version: 1.5.0 on Docker, AMDx64
Adapter hardware: CC2531
Adapter firmware version: Default 20190608

The Logs states that the interview started, and after some time that it was successful:
`zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{\"message\":\"interview_successful\",\"meta\":{\"description\":\"TRADFRI LED bulb GU10 400 lumen, dimmable\",\"friendly_name\":\"0x90fd9ffffe6dd9bb\",\"model\":\"LED1650R5\",\"supported\":true,\"vendor\":\"IKEA\"},\"type\":\"pairing\"}'"}`

Neither the WebUI nor the MQTT topic show the state or brightness channels. It is just `update_available` (Web, MQTT) and `update-state` (MQTT).

I tried that with 5 different bulbs of the same type. all show the same pattern. Since I was not sure if my CC2531 was on its limits, I tried to pair two new Osram Smart+ Plugs (https://www.zigbee2mqtt.io/devices/AB3257001NJ.html), which worked perfect.

Koenkk, 2020-10-23:
The "States" tab is only populated after e.g. the brightness has been set at least once. You can do this via MQTT (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset).

In the latest dev branch an additional tab is added (called "Exposes") which will immediately allow to change these properties. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jti989, 2020-10-26:
Thank you for clarification. I think I had published messages to the /set/state topic using openhab, which got ignored by z2m. Not sure anymore about the exact timely order, I tried so much different things.

Most of my other devices show the states tab correctly direct after pairing, don't know why tradfri behaves other.

Switching to the latest dev branch indeed solved my problem as the states tab then had all that I was searching for.

closed 2020-10-26 by jti989