Koenkk/zigbee2mqtt #4746: zigbee2mqtt 1.15.0 fails to start (Docker, NUC, ConBee II)

info-path, 2020-10-22:
## What happened

Hello, I'm trying to run zigbee2mqtt on a docker, but when it starts, the application stops.

## What did you expect to happen

At least run the application and connect it to the mosquitto-mqtt app.

## How to reproduce it (minimal and precise)

Run app via docker-compose with basic settings.

## Debug info

Zigbee2MQTT version: 1.15.0
Adapter hardware: ConBee II
Adapter firmware: 26660700
Docker host: Intel NUC NUC7i7BNH

## Configuration file
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.30:1883'
  user: <usr>
  password: <pwd>
serial:
  port: /dev/ttyACM0
advanced:
  log_level: debug
```

## The log file:
```
Using '/app/data' as data directory
> zigbee2mqtt@1.15.0 start /app
> node index.js
Zigbee2MQTT:info  2020-10-22 21:26:00: Logging to console and directory: '/app/data/log/2020-10-22.21-26-00' filename: log.txt
Zigbee2MQTT:debug 2020-10-22 21:26:01: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2020-10-22 21:26:01: Starting Zigbee2MQTT version 1.15.0 (commit #1bccc5d)
Zigbee2MQTT:info  2020-10-22 21:26:01: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-22 21:26:01: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
2020-10-22T19:26:01.161Z zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter'
CREATED DECONZ ADAPTER
2020-10-22T19:26:01.162Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[99,98,4,48,238,109,231,48,126,197,222,48,81,100,248,235],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-10-22T19:26:01.162Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2020-10-22T19:26:01.164Z zigbee-herdsman:deconz:driver Serialport opened
2020-10-22T19:26:01.164Z zigbee-herdsman:controller:log Started with result 'resumed'
2020-10-22T19:26:01.164Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2020-10-22T19:26:01.211Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2020-10-22T19:26:11.215Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2020-10-22 21:26:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-22 21:26:11: Failed to start zigbee
Zigbee2MQTT:error 2020-10-22 21:26:11: Exiting...
Zigbee2MQTT:error 2020-10-22 21:26:11: undefined
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
```

Koenkk, 2020-10-23:
@info-path can you provide your herdsman debug logging so @chrishae can look into this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

info-path, 2020-10-23:
@Koenkk I thought I had the right log according to my container envinronment and configuration.yaml settings:

```
...
environment:
      - TZ=Europe/Berlin
      - DEBUG=zigbee-herdsman* npm start
...
```
Do you mean content of this log file:
**/root/.npm/_logs/2020-10-23T23_07_29_404Z-debug.log** ?
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.14.6
3 info using node@v12.18.4
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.15.0~prestart: zigbee2mqtt@1.15.0
6 info lifecycle zigbee2mqtt@1.15.0~start: zigbee2mqtt@1.15.0
7 verbose lifecycle zigbee2mqtt@1.15.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.15.0~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.15.0~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.15.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.15.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.15.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.15.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.15.0
15 verbose cwd /app
16 verbose Linux 5.4.0-51-generic
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v12.18.4
19 verbose npm  v6.14.6
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.15.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.15.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

barnesdannya, 2020-10-24:
I have the same issue using the RaspBeeII running on Raspbian Buster (fully updated including WiringPi) with Zigbee2MQTT 1.15.0-dev. As you can see from the log, I'm also using more recent versions of node and npm than @info-path. I've tried firmware 0x26660700, 0x26650700, and 0x26610700 with the same results.

If you need more information or if you want me to try something with my device, just let me know. :)

**Configuration changes:**
serial:
  port: /dev/ttyAMA0
  adapter: deconz

advanced:
  log_level: debug

**Console:**
> zigbee2mqtt@1.15.0-dev start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2020-10-24 14:30:13: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2020-10-24 14:30:13: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-24.14-30-13' filename: log.txt
Zigbee2MQTT:debug 2020-10-24 14:30:13: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-10-24.03-22-35'
Zigbee2MQTT:info  2020-10-24 14:30:14: Starting Zigbee2MQTT version 1.15.0-dev (commit #bd149af)
Zigbee2MQTT:info  2020-10-24 14:30:14: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-24 14:30:14: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyAMA0"}}'
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2020-10-24 14:30:24: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-24 14:30:24: Failed to start zigbee
Zigbee2MQTT:error 2020-10-24 14:30:24: Exiting...
Zigbee2MQTT:error 2020-10-24 14:30:24: undefined
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-10-24T18_30_24_486Z-debug.log

**Log file:**
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.8
3 info using node@v12.19.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.15.0-dev~prestart: zigbee2mqtt@1.15.0-dev
6 info lifecycle zigbee2mqtt@1.15.0-dev~start: zigbee2mqtt@1.15.0-dev
7 verbose lifecycle zigbee2mqtt@1.15.0-dev~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.15.0-dev~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.15.0-dev~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.15.0-dev~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.15.0-dev~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.15.0-dev~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.15.0-dev start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.15.0-dev
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.4.72-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.19.0
19 verbose npm  v6.14.8
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.15.0-dev start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.15.0-dev start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

ChrisHae, 2020-10-28:
I don't think it is an software problem. Because I'm not aware of these problems from other people or myself even for the last few weeks. Your logs and configuration looks valid. So the error is that the ConBee2 and the RaspBee2 FW does not reply to an simple command (read parameter). My guess would be faulty FW installation. Do you have GCFFlasher installed? It is shipped with the deCONZ software.

Then you could check those commands:
GCFFLasher -l 
this will list your devices and the correct path
GCFFlasher -d /dev/ttyACM0 -r
(use the correct path) This will reset the ConBee2/RaspBee2 and will print the current installed FW Version. Or non if the FW is not installed correctly.

ChrisHae, 2020-10-28:
In rare cases the ConBee/Raspbee can have a faulty configuration (e.g. missing mac address) The easiest way to check this would be to install the deCONZ Software. It has an GUI where you can connect to the ConBee/Raspbee and check all parameters (Network options button at the top middle above the network map) It would be a problem if the MAC address field would be 0x0000000...
You could reconfigure it then.

barnesdannya, 2020-10-28:
@ChrisHae I do have GCFFlasher_internal though I am running a headless build so no GUI. I used:
GCFFlasher_internal -l
and it gave me this path: /dev/ttyS0 instead of /dev/ttyAMA0

After resetting the RaspBee2 and changing the zigbee2mqtt configuration to use /dev/ttyS0, it works perfectly. It is strange though, since I definitely flashed firmware and set up the RTC using /dev/ttyAMA0. What is the difference between these two paths? Is one intended to be used for the RTC and the other for Zigbee (that would make sense in hindsight)?

@info-path Have you tried using the second path, if any?

ChrisHae, 2020-10-29:
The paths are some linux internal thing. I'm not 100% sure about this but I know the paths can differ for some reasons. RaspBee can have /dev/ttyS0 or /dev/ttyAMA0. I think they can also be ttyS1 or AMA1, and so on...

P.S. and of course if deCONZ is installed it must be closed when running zigbee2mqtt.
for the non gui version: 
sudo systemctl status deconz
sudo systemctl stop deconz
sudo systemctl disable deconz
and for the gui version:
sudo systemctl status deconz-gui
sudo systemctl stop deconz-gui
sudo systemctl disable deconz-gui

info-path, 2020-10-29:
> P.S. and of course if deCONZ is installed it must be closed when running zigbee2mqtt.

omg! That's it! I thought that deconz and zigbee2mqtt could be running in parallel independently of each other. After stopping the deCONZ container the zigbee2mqtt started without any problem. I'm very sorry, but this didn't hit me at all.

closed 2020-10-30 by Koenkk

ramon18, 2020-11-07:
I've the exact same as described, in my I'm running it on a Centos 7.
Tried all udev tricks I know, even running as root, always have the error:

`zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1`

The port I'm using
`ls /dev/ttyA*
/dev/ttyACM0`

Also tried the /dev/ttyS0, but it gives a baudrate error.

Any hints?