Koenkk/zigbee2mqtt #4747: XHK1-UE Keypad - Interview failed because can not get node descriptor

Rockman18, 2020-10-22:
Hello

I try to add compatibility with the [XHK1-UE keypad](https://www.ebay.fr/itm/XHK1-UE-Keyboard-Security-System-for-Home-Xfinity-With-4-Batteries/254207297775) but i can't manage to pair the device. Here are the logs + pcap sniff + database.db

I don't have the original gateway to compare the pairing process.

I have  tried the steps in: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails.
I have tried :
- reboot
- directly next to stick
- directly plugged in without extension cable
- replace the batteries of the keypad
- resetting + re-pairing quite a few times
- reflashing the stick, multiple times
- move in another room
- ...

Sometimes in an extremely rare situation the pairing process is better and the device announce itself then nothing more but sorry I can't reproduce that so I don't have logs. 
I hope you can help me to go further

[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/5425983/logs.zip)

Rockman18, 2020-10-26:
@Koenkk : can you take a look and give me an advice on this issue please ?

Rockman18, 2020-10-27:
For the first time since 5 days the pairing go further with the device announcing itself. Sadly my sniffer wasn't plugged in at this time so I don't have the pcapng file. But here is the zigbee-herdsman log file

[pairing_keypad_announce.log](https://github.com/Koenkk/zigbee2mqtt/files/5448575/pairing_keypad_announce.log)

Koenkk, 2020-10-28:
@Rockman18 I've checked your log but the problem is the device does not stay in the network. In the sniff you can see that it keeps rejoining (Association request). I have no clue why, sometimes it helps to compare it with the original gateway but no guarantees here.

Rockman18, 2020-10-28:
So sad. I hope to find a solution soon. Thanks

Rockman18, 2020-11-05:
@Koenkk : I've changed my CC2531 by another one and this time it works...

closed 2020-11-05 by Rockman18