Koenkk/zigbee2mqtt #4748: double messages: action AND click

bartplessers, 2020-10-22:
## What happened
button click on device sends 2 messages:

```
{"action":"single","battery":100,"linkquality":39,"voltage":3022}
{"battery":100,"click":"single","linkquality":39,"voltage":3022}
```

I tried with following settings

```
homeassistant: true
advanced:
  homeassistant_legacy_triggers: **true**
```
this gives

```
Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (65 bytes))
{"action":"single","battery":100,"linkquality":39,"voltage":3022}
Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (64 bytes))
{"battery":100,"click":"single","linkquality":39,"voltage":3022}

Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (59 bytes))
{"action":"","battery":100,"linkquality":39,"voltage":3022}
Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (58 bytes))
{"battery":100,"click":"","linkquality":39,"voltage":3022}

```

Also tested with other settings

```
homeassistant: true
advanced:
  homeassistant_legacy_triggers: **false**
```
  

```
Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (65 bytes))
{"action":"single","battery":100,"linkquality":66,"voltage":3022}
Client mosq-7ZSsaQDV0FZSpwKDbt received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Schakelaar (S10)', ... (64 bytes))
{"battery":100,"click":"single","linkquality":66,"voltage":3022}

```

## What did you expect to happen
Only one message: action OR click, but not both

Koenkk, 2020-10-23:
To disable the legacy integration (`click`) for a device, add the following to the `configuration.yaml` for this device:

```
devices:
  0x12345678:
    friendly_name: my_device
    legacy: false # <-- add this
```

bartplessers, 2020-10-23:
Ok, thanx.
What about that global setting? Is there no other way then modify the setting individually for each device?

Koenkk, 2020-10-24:
to apply to all devices add the following to your configuration.yaml

```
device_options:
  legacy: false
```

closed 2020-10-24 by Koenkk

bartplessers, 2020-10-25:
Hi @Koenkk ,

just one more question:
In the documentaion https://www.zigbee2mqtt.io/information/configuration.html, I could not find this device_option
So now I'm wondering: when was this "legacy: false" introduced?
And more important: will the current message with "action" be stable in the future? I just changed all my automations to use "action" instead of "click"... hopefully this will not change anymore in the future?

kind regards,
Bart

Koenkk, 2020-10-26:
@bartplessers the options for each device individually can also be put in `device_options`, `action` will indeed be stable, click will be removed in zigbee2mqtt 2.0.0