Koenkk/zigbee2mqtt #4750: Adding new device Salus SR600 relay

juranich, 2020-10-23:
## Information about the device + link
Salus SR600 Smart Relay
https://salus-controls.com/uk/product/sr600/

## data/database.db entry of the device
..

Hi,
I’m trying to add Salus SR600, which is not on the list of supported devices.
I’ve followed the instructions https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices
The device is being recognised and there are no errors, however, I cannot seem to get any status changes from it.

I’ve added an entry into devices.js
// SALUS-SR600
{
zigbeeModel: [‘SR600’],
model: ‘SR600’,
vendor: ‘SALUS’,
description: ‘relay switch’,
supports: ‘relay control’,
fromZigbee: [fz.on_off],
toZigbee: [tz.on_off],
meta: {configureKey: 4},
configure: async (device, coordinatorEndpoint) => {
const endpoint = device.getEndpoint(9);
await bind(endpoint, coordinatorEndpoint, [‘genOnOff’]);
await configureReporting.onOff(endpoint);
},
},
I wasn’t sure what to put into “meta” and “configure” so I’ve copied from the already supported “Salus SP600”.

This is the log when I add the device
INFO: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“message”:“interview_successful”,“meta”:{“description”:“relay switch”,“friendly_name”:“0x001e5e090263004d”,“model”:“SR600”,“supported”:true,“vendor”:“SALUS”},“type”:“pairing”}’
INFO: Successfully configured ‘0x001e5e090263004d’

Any help would be greatly appreciated!

Here are some screenshots from Hass.io UI.
[image]
[image]
[image]

Koenkk, 2020-10-23:
Are you able to control the device over MQTT? 

e.g. when sending to `zigbee2mqtt/0x001e5e090263004d/set` payload `{"state":"OFF"}`.

juranich, 2020-10-24:
I've just updated the HassOS to 4.15 and I can now control the device. I didn't do anything else, so the update must have fixed it.

[image]
`INFO: MQTT publish: topic 'zigbee2mqtt/0x001e5e090263004d', payload '{"linkquality":68,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0x001e5e090263004d', payload '{"linkquality":68,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0x001e5e090263004d', payload '{"linkquality":65,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0x001e5e090263004d', payload '{"linkquality":65,"state":"OFF"}'`

However, it still doesn't appear in the Devices or Entities.

Koenkk, 2020-10-24:
Since turning on/off works, I've added support for this device. This should also bring the Home Assistant discovery.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

juranich, 2020-10-26:
Thank you. It woks great!

closed 2020-10-26 by juranich

George-andrew, 2020-11-20:
@Koenkk 
Would you can to add input terminal S1-S2 for connecting physical switch?  

I know that his working at  **"Button Manufacturer-specific" cluster (0xFC08): cluster sever, endpoint:09**
Below table of Attributes SR600

| ID       | Name           | Date Type | M/O/0x1078 | Default | Descriptions |
| ------------- |:-------------:|  -----:| -----:| -----:| -----:|
| 0x0001 | Mode | int8u | O/0x1078 | 0x00 | 0x00 : S1/S2 is non-functional |
| 0x0001 | Mode | int8u | O/0x1078 | 0x00 | 0x01 : S1/S2 will control relay | 
| 0x0001 | Mode | int8u | O/0x1078 | 0x00 | 0x02 : S1/S2 will toggle relay state | 
| 0x0001 | Mode | int8u | O/0x1078 | 0x00 | 0x03 : S1/S2 will trigger state but cannot control relay |
| 0x0001 | Mode | int8u | O/0x1078 | 0x00 | 0x04 : S1/S2 and app will toggle relay status |
| |  |  | |  |  |  |
| 0x0002 | Switch Status | int8u | O/0x1078  | 0x00 | 0x00:S1/S2 open |
| |  |  | |  |  0x01:S1/S2 close |