Koenkk/zigbee2mqtt #4752: SONOFF ZBMINI ZIGBEE TWO WAY SMART SWITCH

taifunk, 2020-10-23:
## Information about the device + link
SONOFF ZBMINI ZIGBEE TWO WAY SMART SWITCH got just released today. Right now it is only available from ITEAD site. I'm allready ordering mine and hope to see it's support  included in the next releases soon :thumbsup:
https://www.itead.cc/sonoff-zbmini-zigbee-smart-switch.html
..

Koenkk, 2020-10-23:
I'm really interested in this device, I've ordered two :smile:

way2many, 2020-10-28:
Ah damn, just bought the Lonsonho 1 gang switch for 17€. This 2 gang switch is 12€.

I hope that a dimming switch will come out soon enough.

Koenkk, 2020-11-06:
@way2many it's not a 2 gang switch, its a 1 gang that can be controlled to way

Device is supported now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-06 by Koenkk

regenbieger, 2020-11-06:
thx for the speed, your faster then the delivery of itead ;)

gcb2018, 2020-11-07:
@Koenkk Thanks! Stupid question but I am not an expert at all: but would the ZBMINI support dimming lights?

I am looking to replace the KAKU [Mini Inbouwdimmer](https://klikaanklikuit.nl/product/mini-inbouwdimmer/)

Koenkk, 2020-11-07:
@gcb2018 no it doesn't support this.

deviantintegral, 2022-02-21:
I've only ever been able to get this device to bind to a single group at a time. I had errors to the effect of `STATUS.TABLE_FULL` with `zha`, and trying this out in zigbee2mqtt it's not working, but I'm also not getting any errors, even with debug logging on.