Koenkk/zigbee2mqtt #4754: Zigbee2mqtt start fails on PI4+CC2538 with USB3 HDD

NikolayMurha, 2020-10-23:
## What happened

zigbee-herdsman start error with timeout

``` 
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8} +261ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:error 2020-10-23 21:38:40: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-23 21:38:40: Failed to start zigbee
Zigbee2MQTT:error 2020-10-23 21:38:40: Exiting...
Zigbee2MQTT:error 2020-10-23 21:38:40: Error: AREQ - ZDO - stateChangeInd after 60000ms
```

## How to reproduce it (minimal and precise)

I try to migrate hassio from PI3 to PI4+HDD. 
During the migration process, I have problems with zigbee2mqtt start.
While researching the problem, I found out that zigbee2mqtt does not start if the USB-HDD is connected to the USB3 port.
I try many combinations CC2538 to usb2 hdd to usb3, both to usb3, both to usb2, etc.
I use two different USB HDD. WD Passort 2TB and jms578 HDD box.

The result is the same. When any USB 3 ports has usb-hdd zigbee2mqtt can't start.
When no HDD on USB3 ports works fine.

Config/Log of success configuration/Log of broken configuration
https://gist.github.com/NikolayMurha/928bae148d0ed575ec1bf2fb64bb120b

## Debug info
Zigbee2MQTT version: 1.15.0 (commit #ed8b4e5)
Adapter hardware: CC2538
Adapter firmware version: ?

oilytin, 2020-10-23:
Having the same issue

oilytin, 2020-10-26:
downgraded cc2531 to CC2531_DEFAULT_20190608.zip (was on 3 originally) and also added start up script. 
have just rebooted the pi and all is working fine.

Anjerlaan, 2020-10-28:
After pi restarts/shutdowns, with multiple usb devices attached, they may have different USB ports. Therefore the 2538 USB Port set in configuration.yaml may have changed, and z2m can’t find the adapter.

Read this: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

closed 2020-12-05 by github-actions[bot]