Koenkk/zigbee2mqtt #4758: Linkquality attribute missing for ikea LED1903C5/LED1835C6

thomasgermain, 2020-10-24:
Hello,

I have a missing attribute `linkquality` in the mqtt payload for kea bulb LED1903C5/LED1835C6, I have 2 of them and I don't have the `linkquality` for both. 

## What happened
The `linkquality` is not sent in the mqtt payload for an Ikea bulb LED1903C5/LED1835C6
 
## What did you expect to happen
The `linkquality` should be in the mqtt payload

## How to reproduce it (minimal and precise)
Pairing a LED1903C5/LED1835C6 bulb and waiting until the payload is sent

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware Conbee II (plugged on a raspberry pi 3b)
Adapter firmware version: 2.05.81 (latest AFAIK)
I'm using the hassio adon (latest version)

Here is the payload for a device with linkquality: https://pastebin.com/LqXHZcb0
And the payload for a LED1903C5/LED1835C6 (linkquality missing): https://pastebin.com/gWtGk8jG

Thanks,

Thomas

Koenkk, 2020-10-24:
This issue has been raised many times (no linkquality for bulbs, e.g. https://github.com/Koenkk/zigbee2mqtt/issues/3353). I always thought it was not possible, but I found a way to get it now.

This is available now in the latest dev branch after executing a command (e.g. turn on) at least twice. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-24 by Koenkk

thomasgermain, 2020-10-24:
Cool 😀 I'll have a try. Thanks !