Koenkk/zigbee2mqtt #4759: Error while starting zigbee-herdsman

McPaHa, 2020-10-24:
Log: https://pastebin.pl/view/42c072a8

## What happened
Error while starting zigbee-herdsman. Tried to reboot Rasberry. Changed usb-port. This was working before but after one night this stopped to start. Tried flash also CC2531_default_20190608.zip

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: ?

HassOS 4.15
Supervisor Version 249

Koenkk, 2020-10-24:
Probably your adapter crashed, your reflash fixed the problem but now you have to delete the `coordinator_backup.json` which can be found in the `data` directory.

McPaHa, 2020-10-25:
Thx... I am little better situation at the moment:

https://pastebin.pl/view/b20b87ca

However all lights are gone/missing? Do I have to pair them again?

Koenkk, 2020-10-25:
yes, requires repairing: https://www.zigbee2mqtt.io/information/FAQ.html#requires-repairing

kpmkpm, 2020-10-28:
I upgraded from cc2531 to cc2652rb with 20200925. Both my cc2531 work ok with my ubuntu20 install, but no luck with this cc2652rb which I reflashed.

`root@ubuntu20:/opt/zigbee2mqtt/data/FLASH# python3 ./test.py
Opening port /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CB_FA_BB-if00-port0, baud 115200

--> Bootloader is working!
--> ZigBee coordinator is working!
`---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20200925

Check done!
`
Logs with debug on, just give this:

`info  2020-10-28 16:30:19: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-28.16-30-19' filename: log.txt
debug 2020-10-28 16:30:19: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2020-10-28 16:30:19: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
info  2020-10-28 16:30:19: Starting zigbee-herdsman...
debug 2020-10-28 16:30:19: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CB_FA_BB-if00-port0"}}'
error 2020-10-28 16:31:34: Error while starting zigbee-herdsman
`

Any ideas?

jbeumer, 2020-11-03:
`Zigbee2MQTT:debug 2020-11-03 20:39:16: Removing old log directory '/share/zigbee2mqtt/log/2020-11-03.20-34-30'
Zigbee2MQTT:info  2020-11-03 20:39:16: Starting Zigbee2MQTT version 1.16.0 (commit #1.16.0)
Zigbee2MQTT:info  2020-11-03 20:39:16: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-11-03 20:39:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"}}'
2020-11-03T20:39:17.238Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-11-03T20:39:17.243Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-11-03T20:39:17.254Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-11-03T20:39:17.273Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-11-03T20:39:17.752Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-11-03T20:39:17.754Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-11-03T20:39:18.770Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:18.783Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-03T20:39:24.794Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:24.796Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-03T20:39:30.804Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:30.807Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2020-11-03 20:39:36: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-03 20:39:36: Failed to start zigbee
Zigbee2MQTT:error 2020-11-03 20:39:36: Exiting...
Zigbee2MQTT:error 2020-11-03 20:39:36: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-03T20_39_36_893Z-debug.log
2020-11-03T20:39:37: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-03T20:39:37: PM2 log: App [npm:0] starting in -fork mode-
2020-11-03T20:39:38: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.0 start /zigbee2mqtt-1.16.0`

closed 2020-12-12 by github-actions[bot]