Koenkk/zigbee2mqtt #4760: State request for Groups

Josov, 2020-10-24:
Currently when requesting the state of a Group using payload = {state: ""} the herdsman replies with the states of each bulb. That's fine however it does not include the state of the Group. If you set the state of the e.g. {state: "OFF"} the herdsman replies with the states of each individual bulb plus the state of the Group. When initialising the system it would be better to request each Group state, retrieve the state of each bulb + the state of the Group and continue from there the processes. This will make the system more transparent.

Koenkk, 2020-10-24:
Zigbee2MQTT currently does not republish the optimistic group state when it did not change. This means the previous published state is still valid. This has been implemented in: https://github.com/Koenkk/zigbee2mqtt/issues/3461

Josov, 2020-10-25:
I understand, but as far as I know it is not possible to inquire the actual state of the Group. The only way to get the Group state is to set the Group and the herdsman replies (including the Group state).

closed 2020-12-02 by github-actions[bot]