Koenkk/zigbee2mqtt #4764: wrong device

msimko81, 2020-10-24:
## What happened
Unable to start ZigBee2MQTT in docker container. Logs from container: https://pastebin.com/0ArU9fDL

## What did you expect to happen
Container is started without any error.

## How to reproduce it (minimal and precise)
`docker run -it -v $(pwd)/data:/app/data --device=/dev/ttyUSB0 -e TZ=Europe/Prague -v /run/udev:/run/udev:ro --privileged=true koenkk/zigbee2mqtt`

The device is listening on /dev/ttyUSB0, user miro is in group dialout and can access the device. 

## Debug info
Zigbee2MQTT version: koenkk/zigbee2mqtt:latest, image ID: f410c085c356
Adapter hardware: CC26X2R1
Adapter firmware version: Z-Stack 3.x.0 20200925 firmware

Koenkk, 2020-10-24:
```
Zigbee2MQTT:error 2020-10-24 16:55:43: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
```

In your post you mention `/dev/ttyUSB0`, that means you did not specify the correct port in your `configuration.yaml`. See the `serial` part of https://www.zigbee2mqtt.io/information/configuration.html

msimko81, 2020-10-24:
Thank you! Now I am able to start it.

closed 2020-10-24 by msimko81