Koenkk/zigbee2mqtt #4765: Develco ZHOT101 motion sensor support

Gelisob, 2020-10-24:
Hello, after following how to support new devices guides i've figured, that what I attempt to do is out of my abilities at the moment. 
Since the devices fail their interview and I get no model info, the support must be added from deeper in the addon?
They do get listed in the addon home view, with manuf and model "unsupported".

## Information about the device + link
I have found several Develco ZHOT101 motion sensors that I would like to not throw away.
https://wattle.com/support/our-products/older-products/motion-sensors/hc-motion-sensor/
Stripped board and chip view pictures, if it matters: 
https://imgur.com/a/X9KHDCA

I'm running plain hass.io image install version 0.116.2
I didnt find information on how to get herdsman log file out of hass.io install after enabling the debugging flags in conf. The log files in log folder dont get debug level data.

I gathered what I could from the log window within addon, please let me know if it's not all that is required.
[herdsmanLog-0015bc001a003270.txt](https://github.com/Koenkk/zigbee2mqtt/files/5433786/herdsmanLog-0015bc001a003270.txt)

Or if i can actually add support by myself somehow that i wasnt able to find information about?

Gelisob, 2020-10-25:
I'm guessing it should report five things, by the webpage information.
1. movement
2. lux
3. temperature
4. battery
5. signal strength

I took it apart to see if i can find temp sensor but didnt see obvious one, maybe it's inside some other sensor? Motion and lux sensors were obvious.

Gelisob, 2020-10-25:
hmm, looking at other new device support requests, i saw someone posting database.db entry and went looking at it, this actually does show both manufacturer and model.

This is the entry for the motion sensor:
{"id":2,"type":"EndDevice","ieeeAddr":"0x0015bc001a003270","nwkAddr":1107,"manufId":0,"manufName":"Develco","powerSource":"Battery","modelId":"ZHOT201","epList":[1,34,35,38,39,40,41],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[2,3,5],"outClusterList":[1],"clusters":{},"binds":[],"meta":{}},"34":{"profId":260,"epId":34,"devId":263,"inClusterList":[0,3,1,1030],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"zoneState":0}}},"binds":[],"meta":{}},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1,1026],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"39":{"profId":260,"epId":39,"devId":262,"inClusterList":[0,3,1,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"40":{"profId":260,"epId":40,"devId":263,"inClusterList":[0,3,1,1030],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"41":{"profId":260,"epId":41,"devId":263,"inClusterList":[0,3,1,1030],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}}},"zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1603573745626}

Gelisob, 2020-10-25:
With this information, i made devices.js entry

	{
    zigbeeModel: ['ZHOT201'], 
    model: 'ZHOT201', 
    vendor: 'Develco', 
    description: 'Motion, lux & temperature sensor', 
    supports: 'occupancy, temperature, lux and battery', 
    fromZigbee: [], 
    toZigbee: [], 
	},

This resulted in device model and manufacturer being show, 
[image]

but interview still fails.

After that it just spams a device_announced broadcast. Don't really know what should be the next step.

Here are interview and herdsman log parts:

INFO: Starting interview of '0x0015bc001a003270'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0015bc001a003270"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a003270"},"type":"device_announced"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a003270"},"type":"device_announced"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a003270"},"type":"device_announced"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a003270"},"type":"device_announced"}'
ERROR: Failed to interview '0x0015bc001a003270', device has not successfully been paired
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270","status":"failed"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b0018e306f1","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"definition":{"description":"Motion, lux & temperature sensor","model":"ZHOT201","supports":"occupancy, temperature, lux and battery","vendor":"Develco"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genDeviceTempCfg","genIdentify","genScenes"],"output":["genPowerCfg"]}},"34":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","msOccupancySensing"],"output":["genTime"]}},"35":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","ssIasZone"],"output":[]}},"38":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","msTemperatureMeasurement"],"output":[]}},"39":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","msIlluminanceMeasurement"],"output":[]}},"40":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","msOccupancySensing"],"output":["genTime"]}},"41":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genPowerCfg","msOccupancySensing"],"output":["genTime"]}}},"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270","interview_completed":false,"interviewing":false,"network_address":1107,"power_source":"Battery","supported":true,"type":"EndDevice"}]'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x0015bc001a003270"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a003270"},"type":"device_announced"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a003270","ieee_address":"0x0015bc001a003270"},"type":"device_announce"}'
[... cut ...]
2020-10-25T19:15:45.993Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,170,159]
2020-10-25T19:15:45.994Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,170,159]
2020-10-25T19:15:45.994Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,170] - 159
2020-10-25T19:15:45.994Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":170}
2020-10-25T19:15:45.994Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Koenkk, 2020-10-26:
Can you provide the complete herdsman debug log of the pair? (dont post it here but on pastebin.com, and provide link here
)

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Gelisob, 2020-10-26:
> Can you provide the complete herdsman debug log of the pair? (dont post it here but on pastebin.com, and provide link here
> )
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Sure, but like i described before, this guide doesn't describe how to get the herdsman debug log out of hass.io

The \share\zigbee2mqtt\log folder files do not include herdsman debug output. 
I only can press refresh within the addon "log" menu, but this doesnt have many lines and i cant copy-paste fast enough not to miss a lot of it :)

Suggestions?

I havent found hass.io addon writing herdsman log to anywhere.

Gelisob, 2020-10-26:
Okay the workaround I came up with was to copy my addon "Log" tab browser address:
http://yourhass.io:8123/hassio/addon/<usewhatyourshas>_zigbee2mqtt/logs
Start pairing, refresh current tab, open new tab, paste url, enter, new tab.. etc, every two seconds and then copy paste logs together.
This still left holes in the log, after several attempts, even using 3 monitors with 6 tabs pre-opened to just press "refresh" in the hass.io ui, i still kept getting holes in the log. Specially in the beginning. Refer to this method if hass.io doesnt have better options right now.

Gelisob, 2020-10-26:
Here's my spotty log, please take a look at it if this has enough information @Koenkk . I'm not sure if the ui refresh method can be used to get the big data junks. Do let me know if there's too much missing, thanks!

[herdsmanLog-spotty271020-0130.txt](https://github.com/Koenkk/zigbee2mqtt/files/5442082/herdsmanLog-spotty271020-0130.txt)

Koenkk, 2020-10-28:
Looks that the device does not correctly follow the zigbee spec, it returns an unsupported attribute on the zoneState attribute, however this attribute is mandatory. That is the reason why interviewing fails, nevertheless there is still a chance we can make the device work. 

With the following definition:

```js
{
    zigbeeModel: ['ZHOT201'],
    model: 'ZHOT201',
    vendor: 'Develco',
    description: 'Motion, lux & temperature sensor',
    supports: 'occupancy, temperature, lux and battery',
    fromZigbee: [],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpiont(35);
        await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
        const payload = {enrollrspcode: 0, zoneid: 23};
        await endpoint.command('ssIasZone', 'enrollRsp', payload, {disableDefaultResponse: true});
    },
},
```

- Repair the device
- Check if you see `Succesfully configured..` for this device after pairing it
- Do you get any `No converter available..` messages for this device when triggering occupancy?

Gelisob, 2020-10-28:
Hey thanks for the input @Koenkk , these are my first zigbee devices so sorry for excess or too little information. Still in progress to understand how this system and sensors walk and talk.

Looks like progress but the interview was not success, I picked together some of the log outputs what I thought seem relevant.
Am I off from correct devices.js skew? Seemed to fail with "TypeError: device.getEndpiont is not a function" at devices.js
pointing at line 12837, thats 
"const endpoint = device.getEndpiont(35);" 
for me, in the code snipped you made.

[herdsmanLog-selective281020.txt](https://github.com/Koenkk/zigbee2mqtt/files/5455034/herdsmanLog-selective281020.txt)

So this time after unsuccess pairing i opened the-not-paired-but-somehow-still-paired? state device and here's a pic of it's endpoints. I'm guessing these are somehow the sensor frame reports or.. not sure how to think about them.
(https://user-images.githubusercontent.com/22873920/97501227-9e269d00-1979-11eb-849a-38754ec6cfc3.jpg)

In the log you can also see some "no converter available" popping up now.

Let me know if I should gather-hunt some herdsman log info around certain endpoints or something.

Koenkk, 2020-10-29:
I made a typo, in https://github.com/Koenkk/zigbee2mqtt/issues/4765#issuecomment-718045340 change `const endpoint = device.getEndpiont(35);` to `const endpoint = device.getEndpoint(35);`

Gelisob, 2020-10-30:
Doh, and I was staring right at it and didnt notice too, shame on me.

Now, with typo fixed, it still doesnt succeed with interview. No mention of "Succesfully" in the logs.
I grabbed some log for you to look at. Got no converter messages and picked some of the clusters I could find.
cluster 'genBasic' 
cluster 'ssIasZone' 

adding log file
[herdsmanLog-selective301020.txt](https://github.com/Koenkk/zigbee2mqtt/files/5467834/herdsmanLog-selective301020.txt)

2020-10-30T21:43:39.151Z zigbee-herdsman:controller:error Interview failed for '0x0015bc001a003270 with error 'Error: Read 0x0015bc001a003270/35 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2020-10-30 23:43:39: Failed to interview '0x0015bc001a003270', device has not successfully been paired

After that, keeps the announce looping like before for some time.

Koenkk, 2020-10-31:
After the interview, can you wakeup the device and force a configure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Gelisob, 2020-10-31:
Looks like I can, here's the resulting communication: 
[herdsmanLog-configureTriggerMQTT-311020.txt](https://github.com/Koenkk/zigbee2mqtt/files/5468882/herdsmanLog-configureTriggerMQTT-311020.txt)

Koenkk, 2020-10-31:
`2020-10-31T10:17:22.019Z zigbee-herdsman:controller:endpoint Write 0x0015bc001a003270/35 ssIasZone({"iasCieAddr":"0x00124b0018e306f1"}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))`  indicated the device is sleeping , make sure to wake it up right before executing the command by clicking a button on it.

Gelisob, 2020-11-01:
Hmm, having difficulty getting it into more talkative state. Here's one attempt, it ended with mac transaction expiration too but in the beginning the ssIasZone endpoint write call did answer.. please take a look if this log contains more helpful communication
[herdsmanLog-configureTriggerMQTT-011120.txt](https://github.com/Koenkk/zigbee2mqtt/files/5470333/herdsmanLog-configureTriggerMQTT-011120.txt)

Koenkk, 2020-11-01:
No luck unfortunately, as this device shows non-standard behaviour I'm afraid you have to sniff traffic with the original gateway in order to figure out how it works there. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Gelisob, 2020-11-02:
Thank you @Koenkk for time and effort, trying to make these work. 
Unfortunately it makes too little sense to buy ~100 eur gateway (that also is not guaranteed to support it) to continue this research for the 5 old motion sensors. Specially if sonoff zigbees are under 10$ already. Tho these are essentially multisensors..

I'll let them collect dust in cellar for some years maybe and if future brings new developments that might help make them work (maybe someone with wattle gateway sniffs something so you can apply it) - tag and reopen :)

closed 2020-11-02 by Gelisob