Koenkk/zigbee2mqtt #4769: Inconsistent return values when getting hue and saturation

TilmanK, 2020-10-25:
## What happened
When requesting hue and saturation from my Philips Hue devices I get color back as x and y and color temperature.

## What did you expect to happen
Z2M requests color properties and reports them back as requested (hue + saturation). No color temperature is returned.

## How to reproduce it (minimal and precise)
Set a JSON payload to get color to /get topic:

{
  'color': {
    'hue': null,
    'saturation': null
  }
}

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version:

Koenkk, 2020-10-27:
Fixed, please verify.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

TilmanK, 2020-10-27:
Thanks, Koen. Works like a charm!

closed 2020-10-27 by TilmanK