Koenkk/zigbee2mqtt #4771: Constantly increasing delay when using Osram RGBW led strip with TV Ambilight (diyHue bridge) and HA

Mr-Groch, 2020-10-25:
## What happened
I'm using about 20 devices zigbee network on rpi3b+ and hassos with success. I have Philips TV with amibilight and OSRMAM RGBW led strip on my zigbee network, so I wanted to use it for Ambilight+Hue with my TV. I've installed diyHue bridge on docker container and I've added to it my LED strip (diyHue is connected to same Mosquito MQTT broker as Zigbee2MQTT and HA and uses zgbee2mqtt set messages to control LED strip). Ambilight+Hue configuration was success and my Philips TV is able to control LED strip. 

Problem is that after a short time of using Ambilight+Hue, LED strip reaction is becoming more and more laggy. Delay of setting LED color is constantly increasing. After turning Ambilight+Hue off, LED strip is still changing colors. And when I try to switch it off, it will be turned off after minute or two... Controlling other devices in my zigbee network at the same time is fine - switching ON/OFF other lights from HA is almost instant.

I see in zignee2mqtt log that set messages for LED strip is published even 10x in 1 second, but also there are few breaks lasting few second, without publishing state messages...
Logs: http://wklejto.pl/864016

Also I see in HA database, really huge number if state chaanges event for LED strip device, this is total count of state change events for LED strip compared to other entity state change events ordered by count:

entity_id | count(*)
-- | --
light.tasma_led_kolorowa_zewnetrzna_1_light | 12105
sensor.tasma_led_kolorowa_zewnetrzna_1_linkquality | 7603
sensor.tasma_led_kolorowa_zewnetrzna_1_update_stat... | 7598
binary_sensor.tasma_led_kolorowa_zewnetrzna_1_upda... | 7594
sensor.magic_cube_1_action | 457
sensor.magic_cube_1_linkquality | 329
binary_sensor.drzwi_kuchnia_contact | 284
sensor.drzwi_kuchnia_linkquality | 270

## What did you expect to happen
I think that problem is that every set messages is queued, and device (coordinator? rpi? LED strip?) can't handle it instantly, so the solution should be:
- set topic messages should not be queued, but dropped if they can't be handled instantly (maybe configurable?)
- configurable max handled messages in some interval for device (messages over limit should be ignored)
- possibility to temporary disable publishing state messages from zigbee2mqtt to HA for device (something like external control mode enabled/disabled)
- other solution?

## How to reproduce it (minimal and precise)
1. Connect to same mqtt broker zigbee2mqtt, HA and diyHue
2. Pair Philips TV to diyHue and configure RGBW light for use with Ambilight+Hue function
3. Enable Ambilight+Hue function on TV and observe incresing lagg in controlling RGBW light.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 20190608

Mr-Groch, 2020-10-28:
Look at this debug log - last received MQTT message for changing LED strip light was received at 17:02:18, but Zigbee2MQTT continue to publish Zigbee and MQTT messages for this LED strip 2 minutes after that (last Zigbee set message transmited at 17:04:10)

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5453249/log.txt)

Koenkk, 2020-10-29:
Currently there is no option to do some throttling, I consider this as a feature request rather then a problem. 

To improve the throughput you could disable the default response for your device in devices.js (add `meta: { disableDefaultResponse: true},` like here https://github.com/Koenkk/zigbee-herdsman-converters/blob/1fee3ec22c6dbb12e55626ec3152a86d0c32a732/devices.js#L4822

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

Mr-Groch, 2020-10-29:
Thanks for the reply. Unfortunately it didn't help much... I have modified device like that:

```
    {
        zigbeeModel: ['LIGHTIFY Outdoor Flex RGBW', 'LIGHTIFY FLEX OUTDOOR RGBW', 'Flex Outdoor RGBW'],
        model: '4058075036185',
        vendor: 'OSRAM',
        description: 'Outdoor Flex RGBW test',
        extend: ledvance.light_onoff_brightness_colortemp_colorxy,
        meta: {disableDefaultResponse: true},
        ota: ota.ledvance,
    },
```

Also added `test` string to end of `description` to verify in frontend, that my modified devices.js works...

LED strip delay still is constantly increasing when using Ambilight. And still so much mqtt messages is published by zigbee2mqtt and recorded by homeassistant.

Mr-Groch, 2020-11-29:
Request is still valid.

svrooij, 2021-01-06:
If developed the initial support for mqtt (zigbee2mqtt) in diyHue and when I did I saw that there was room for improvement on the entertainment (hue sync box) field.

Now it tries to send messages to each device, instead of how the official hue bridge does this. The hue bridge groups the lights that should receive the same message and then only sends one message to each group.

but this seems like that the rate limit should be implemented on diyhue and not in zigbee2mqtt.

mariusmotea, 2021-01-14:
@Koenkk  it will be grat if we can stream the entertainment frames in realtime. For wifi lights we implemented dedicated UDP channel and even if there are short communication delays in the network, lights will always display the last sent state and this made them to operate like original Hue lights. With zigbee2mqtt currently i need to fine tune a divisor to drop some frames from original stream (24 frames / second). It seams that there is a caching mechanism on the path to the light, if this can be disabled it may work much better.

mariusmotea, 2021-02-14:
No stale, is a feature request.

closed 2021-03-24 by github-actions[bot]