Koenkk/zigbee2mqtt #4774: CC2538 Router (Range extender)

Xsev1, 2020-10-25:
## Information about the device + link
..Jethome Router Firmware for cc2538 chips (modkam ru) https://github.com/jethome-ru/zigbee-firmware/tree/master/ti

## data/database.db entry of the device
..
model:cc2538.router.v1
type:Router
nwk:49795
manuf id:0
manufacturer:jethome
power:Mains (single phase)
hard version:1
zcl version:1
date code:20200522
interviewed:true
configured:false
endpoint:1
profile:260
input clusters:genBasic (0)
genIdentify (3)
output clusters:genBasic (0)

Koenkk, 2020-10-28:
Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html and create a pr for https://github.com/Koenkk/zigbee-herdsman-converters

closed 2020-10-28 by Koenkk

dzikus, 2020-11-18:
> Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html and create a pr for https://github.com/Koenkk/zigbee-herdsman-converters

I did exactly what is described in the link above.

ad.1 I paired my cc2538+cc2592 router.
ad.2 I added entry to node_modules/zigbee-herdsman-converters/devices.js
ad.3 Nothing to do, router do not need converters.
ad.4 ignored as optional
ad.5 ignored as optional
ad.6 done

So how can we follow your guide and have cc2538 as router if in your instruction there is no merge of our code with yours and you do not want to add id by yourself.

Be consistent and add a decent manual as you refer people to it, dismissing them!