Koenkk/zigbee2mqtt #4775: Tuya door/window sensor cannot be connected.

bennietrinks, 2020-10-25:
I've been busy for days, but can't find a solution.
Tuya door / window sensor, the printed circuit board says MC100-V0.3
This is a Zigbee door / window sensor from the brand Tuya.

According to the manual, I have to hold the button on it then the light should turn green and when the green light is off I have to release the button and start flashing green.

Everything goes well and all according to the manual, but in Zigbee2mqtt I don't see anything coming in while permit join is on.

To register a new device you must have debugging information.
They state there that every Zigbee device must be able to connect, but this does not seem to be OK.
Have four of those sensors, all have the same problem.

Does anyone have an idea?

[image]

dinhchinh82, 2020-10-25:
Can you provide the debug log file?
https://www.zigbee2mqtt.io/information/debug.html

bennietrinks, 2020-10-25:
Thank you for your quick response. Here is my log when I have debug on.

https://pastebin.pl/view/387eafd3

bennietrinks, 2020-10-27:
Did I provide the correct information? Do you need more information? Is there a solution for this? Viewed everything on the Internet, but don't go any further.

closed 2020-12-04 by github-actions[bot]

mooney40, 2022-11-26:
1,Open up remove circuit board 
2,now look for the button to press and hold to get LED to flash for binding to zigbee hub
3, push button and hold for five seconds  if it dose not start flashing
4, hold metal conductive material to short the two connections were button is soldred to circuitboard for five seconds or until LED flashes I have had faulty buttons on mine
5, if still no flashing LED then check battery they sometimes come DOA
6, battery holder often needs adjustment as they need to pinch battery to make good contact.
7, if all done then start looking at software. 
Sorry can't help with that part!
Just a quick check you are using a hub
To connect it