Koenkk/zigbee2mqtt #4776: Zemismart light switch not recognised properly

passy05, 2020-10-25:
I have a Zemismart light switch identical to https://www.zigbee2mqtt.io/devices/TS0043.html. It is working fine in Home Assistant but I would like to be able to use the action attribute but I can't because it is detected as https://www.zigbee2mqtt.io/devices/TS0013.html:
Which debugging information is needed for this device (I am using Home assistant add-on)?

Koenkk, 2020-10-26:
Can you provide the `data/database.db` entry of your device?

passy05, 2020-10-26:
Thanks, here it is:

{"id":74,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe687f3f","nwkAddr":41593,"manufId":4098,"manufName":"_TYZB01_gvackgwu","powerSource":"Battery","modelId":"TS0013","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0013","manufacturerName":"_TYZB01_gvackgwu","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00219ee54f","endpointID":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1603519574338}

Koenkk, 2020-10-26:
This device announces itself as a `TS0013`, can you provide the debug logging when clicking the buttons?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

passy05, 2020-10-27:
Here is the log after I switch on and off 10s later:

debug 2020-10-27 08:34:03: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:03: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":39,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
info  2020-10-27 08:34:03: Configuring 'KitchenLights'
debug 2020-10-27 08:34:14: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:14: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":42,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
debug 2020-10-27 08:34:14: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:14: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":39,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
debug 2020-10-27 08:34:15: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:15: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":39,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
debug 2020-10-27 08:34:15: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:15: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":39,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
debug 2020-10-27 08:34:15: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:15: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":42,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
debug 2020-10-27 08:34:16: Received Zigbee message from 'KitchenLights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-10-27 08:34:16: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":39,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
error 2020-10-27 08:34:27: Failed to configure 'KitchenLights', attempt 3 (Error: ConfigureReporting 0xec1bbdfffe687f3f/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 41593 - 1 - 34 - 6 - 7 after 10000ms)

Koenkk, 2020-10-28:
@passy05 looks that this is not a TS0043 as this provides the events in a different way (note that also the picture on the docs can be a bit of). 

- TS0043 has no relays (you cannot wire a lamp to it). It only provides actions, you have to see it like a remote.
- TS0013 has relais

passy05, 2020-10-28:
Thanks a lot for your help! I realised now that I was misled by the pictures and effectively the device description of the TS0013 matches mine.

closed 2020-10-28 by passy05