Koenkk/zigbee2mqtt #4778: Trying To Add New Device Support But Device Detected As Empty String

danieledwardgeorgehitchcock, 2020-10-26:
## What happened
I am trying to pair a Yale Connexis L1 Zigbee module (manufactured and supplied by Control4) - I have managed to get it to successfully interview and link with my network.

## What did you expect to happen
I viewed the debug log to then be able to add support for the device but, there are no attributes being returned, just empty strings (i.e. zigbee model):

```
Zigbee2MQTT:info  2020-10-26 09:14:58: Device '0x000d6f0011141333' joined

Zigbee2MQTT:info  2020-10-26 09:14:58: Starting interview of '0x000d6f0011141333'

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0011141333","ieee_address":"0x000d6f0011141333"},"type":"device_joined"}'

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0011141333","ieee_address":"0x000d6f0011141333","status":"started"},"type":"device_interview"}'

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f0011141333"},"type":"device_connected"}'

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f0011141333"},"type":"pairing"}'

Zigbee2MQTT:debug 2020-10-26 09:14:58: Device '0x000d6f0011141333' announced itself

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0011141333","ieee_address":"0x000d6f0011141333"},"type":"device_announce"}'

Zigbee2MQTT:info  2020-10-26 09:14:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x000d6f0011141333"},"type":"device_announced"}'

Zigbee2MQTT:debug 2020-10-26 09:15:13: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"modelId":""}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:14: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Yale"}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:15: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:15: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:16: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"appVersion":33}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:17: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:18: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{"hwVersion":15}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:18: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-10-26 09:15:19: Received Zigbee message from '0x000d6f0011141333', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2020-10-26 09:15:20: Successfully interviewed '0x000d6f0011141333', device has successfully been paired

Zigbee2MQTT:warn  2020-10-26 09:15:20: Device '0x000d6f0011141333' with Zigbee model '' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Zigbee2MQTT:info  2020-10-26 09:15:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x000d6f0011141333","ieee_address":"0x000d6f0011141333","status":"successful","supported":false},"type":"device_interview"}'
```

I believe the Control4 Zigbee protocol is yet another deviation away from the standard Zigbee protocol (they call it Control4 Zigbee Pro) [Here is a general manual on the topic](https://www.control4.com/docs/product/zigbee/best-practices/english/latest/zigbee-best-practices-rev-d.pdf#:~:text=%2Fpcna%2Dfaq.-,Control4%20Zigbee%20PRO,in%20sophisticated%20smart%2Dhome%20systems.)

I hope that it doesn't deviate enough that these modules can't be used - I just require a bit more guidance (this is only my second device support addition)

## How to reproduce it (minimal and precise)
Insert Zigbee module into Yale Conexis L1 door lock
Press R button 3 times to enter pairing mode
Permit join on Zigbee2MQTT
Wait for door lock to chime and confirm in Zigbee2MQTT that device has paired
Review Zigbee2MQTT logs

## Debug info
Zigbee2MQTT version: 1.15.0-dev
Adapter hardware: CC26X2R1
Adapter firmware version: 20200805

danieledwardgeorgehitchcock, 2020-10-26:
So I have got so far with this config but, the state / reporting doesn't seem to work:

```
const device = {
    zigbeeModel: [''],
    model: 'ZYA-C4-MOD-S',
    vendor: 'Yale',
    description: 'Control4 module for Yale KeyFree/Keyless lock',
    supports: 'lock/unlock, battery',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
    toZigbee: [tz.generic_lock],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await configureReporting.lockState(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.lock(), e.battery()],
};
```

Also, I am not sure I can push this to production when I have managed to get it working due to the empty zigbeeModel string.

Can anyone help me get this set up better? It seems the configure flow times out...

danieledwardgeorgehitchcock, 2020-10-26:
Getting these errors when pairing:

```
Zigbee2MQTT:error 2020-10-26 14:20:18: Failed to configure '0x000d6f0011141333', attempt 1 (Error: ConfigureReporting 0x000d6f0011141333/1 closuresDoorLock([{"attribute":"lockState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 44064 - 1 - 13 - 257 - 7 after 10000ms)
```

And...

```
Zigbee2MQTT:debug 2020-10-26 14:19:52: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":""}'

Zigbee2MQTT:debug 2020-10-26 14:19:52: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"Yale"}'

Zigbee2MQTT:debug 2020-10-26 14:19:52: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'

Zigbee2MQTT:debug 2020-10-26 14:19:52: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'

Zigbee2MQTT:debug 2020-10-26 14:19:53: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":33}'

Zigbee2MQTT:debug 2020-10-26 14:19:53: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{}'

Zigbee2MQTT:debug 2020-10-26 14:19:53: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":15}'

Zigbee2MQTT:debug 2020-10-26 14:19:54: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{}'

Zigbee2MQTT:debug 2020-10-26 14:19:54: No converter available for 'ZYA-C4-MOD-S' with cluster 'genBasic' and type 'readResponse' and data '{}'
```

Koenkk, 2020-10-26:
Can you provide the `data/database.db` entry of this device?

danieledwardgeorgehitchcock, 2020-10-26:
```
{"id":27,"type":"EndDevice","ieeeAddr":"0x000d6f0011141333","nwkAddr":44064,"manufId":43690,"manufName":"Yale","powerSource":"Battery","modelId":"","epList":[1,196],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,9,10,257,64512,1],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"","manufacturerName":"Yale","powerSource":3,"zclVersion":1,"appVersion":33,"hwVersion":15}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004f1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004f1","endpointID":1}],"meta":{}},"196":{"profId":260,"epId":196,"devId":10,"inClusterList":[1],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"appVersion":33,"hwVersion":15,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1603721995280}
```

Koenkk, 2020-10-26:
@danieledwardgeorgehitchcock instead of `zigbeeModel` use this:

```
fingerprint: [
    {type: 'EndDevice', manufacturerName: 'Yale', manufacturerID: 43690, powerSource: 'Battery', endpoints: [
        {ID: 1, profileID: 260, deviceID: 10, inputClusters: [0,9,10,257,64512,1], outputClusters: []},
        {ID: 1, profileID: 260, deviceID: 10, inputClusters: [1], outputClusters: []},
    ]},
],
```

danieledwardgeorgehitchcock, 2020-10-27:
Hi @Koenkk 

I tried the above and it didnt work. I also made some slight amendments to fit the device profile in the database. my device now looks like this:

```
    {
        fingerprint: [{
            type: 'EndDevice',
            manufacturerName: 'Yale',
            manufacturerID: 43690,
            powerSource: 'Battery',
            endpoints: [
               {ID: 1, profileID: 260, deviceID: 10, inputClusters: [0,9,10,257,64512,1], outputClusters: []},
               {ID: 196, profileID: 260, deviceID: 10, inputClusters: [1], outputClusters: []},
            ],
        }],
        model: 'ZYA-C4-MOD-S',
        vendor: 'Yale',
        description: 'Control4 module for Yale KeyFree/Keyless lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.generic_lock],
        meta: {configureKey: 1, timeout: 50000},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.lock(), e.battery()],
    }
```

However, I am back to getting the standard message:

```
Zigbee2MQTT:info  2020-10-27 08:10:29: Successfully interviewed '0x000d6f0011141333', device has successfully been paired

Zigbee2MQTT:warn  2020-10-27 08:10:29: Device '0x000d6f0011141333' with Zigbee model '' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Zigbee2MQTT:info  2020-10-27 08:10:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x000d6f0011141333","ieee_address":"0x000d6f0011141333","status":"successful","supported":false},"type":"device_interview"}'
```

danieledwardgeorgehitchcock, 2020-10-27:
I have just done a Zigbee sniff of this device joining my network... As feared, it would appear that Control4 has added another layer on top of the Zigbee protocol. This is where my knowledge comes to an end!

[See this unfiltered PCAP file](https://drive.google.com/file/d/1zOAEORm_2q2iNX37asKQeStN98jhnVdA/view?usp=sharing)

Koenkk, 2020-10-28:
I didn't know about the C4 Zigbee protocol until now. But are you able to lock/unlock the lock by sending to `zigbee2mqtt/0x000d6f0011141333/set` payload `{"state": "LOCK"}`?

danieledwardgeorgehitchcock, 2020-10-29:
I didn't know either.

I can unlock the door by sending a payload of `{"state": "UNLOCK"}` but, sending a a payload of `{"state": "LOCK"}` after results in the following error:

```
Zigbee2MQTT:info  2020-10-29 08:43:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000d6f0011141333' failed: 'Error: Command 0x000d6f0011141333/1 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 15471 - 1 - 21 - 257 - 0 after 10000ms)'","meta":{"friendly_name":"0x000d6f0011141333"},"type":"zigbee_publish_error"}'
```

In the new Zigbee2MQTT GUI, I don't see anything in the States tab for this device other than last_seen, but the Exposes tab is populated with the lock state (with a greyed out switch and LOCK / UNLOCK either side of it), battery (which just shows N/A%) and LQI (which also shows NA%). I can confirm that I can also unlock the device through the Exposes tab, but lock gives the same error as above.

I don't think the device is correctly configured in Zigbee2MQTT which may be leading to this sort of behaviour?

Trouble is, I don't have any C4 gear to be able to sniff the device during normal operation :-(

Koenkk, 2020-10-29:
`Zigbee2MQTT:info  2020-10-29 10:47:36: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x000d6f0011141333"},"error":"Failed to configure (endpoint is not defined)","status":"error"}'
` is interesting. Can you provide the Zigbee2MQTT debug log when executing the configure?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

danieledwardgeorgehitchcock, 2020-10-30:
Ok, I cannot seem to replicate the previous behaviour now - My guess is that there was some sort of timeout between Zigbee2MQTT and the device.

Trouble is, I still have nothing showing in the States tab and the battery status and LQI are still showing as N/A% but it seems to respond to set commands... Guessing there is something not right with the config / reporting on initial setup..?

I will try and capture a full interview and config in Zigbee2MQTT and Wireshark later...

theperm, 2021-03-10:
Why did this close of its a feature request. I too am looking for zigbee control of this lock. Send this control 4 module is the only option. Did the OP give up?

danieledwardgeorgehitchcock, 2021-03-10:
I’m still here! Unfortunately, the interface connector on my Conexis lock broke while I was developing this and while I have a replacement, I am currently in the process of moving home.

the code above works to a point if you add it as a custom extension - it’s just the reporting that was a bit squiffy!

elgerg, 2021-08-05:
> Why did this close if its a feature request. I too am looking for zigbee control of this lock. Seems this control 4 module is the only option. Did the OP give up?

I have one of these that is controlled by ZWave. Id much rather it be done by Zigbee...

danieledwardgeorgehitchcock, 2021-08-06:
I didn’t give up, I moved home and am yet to fit my lock to be able to re-open the development :-)

Coedy, 2021-11-17:
Subscribing to this - I'd happily pay the *gulp* £90 for the zigbee interface on top of the lock cost, if I could get the lock working with HA via Z2Mqtt...

Haven't bought it yet as still not "officially supported" but thank you for your valiant efforts so far!

danieledwardgeorgehitchcock, 2021-11-17:
Hi @Coedy,

You might want to subscribe to this one too as I have reason to believe that the Doorman and the Conexis modules may be compatible: Koenkk/zigbee-herdsman-converters#3326

ChipWolf, 2021-11-18:
I just bought this module & am also looking to use it with zigbee2mqtt

ChipWolf, 2021-11-18:
I just bought this module & am also looking to use it with zigbee2mqtt
@Koenkk, please can we re-open this?

danieledwardgeorgehitchcock, 2021-11-19:
Ok,

As this has now been re-opened, I may as well share my progress.

I moved house and have only just got around to re-fitting the lock with this module.

I have purchased a Control4 Zigbee Hub (extremely expensive!!) and am in the process of getting it all set up so that I can sniff the traffic.

For now, here is the external converter that allows me to lock and unlock the door without reporting:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
        fingerprint: [{
            type: 'EndDevice',
            manufacturerName: 'Yale',
            manufacturerID: 43690,
            powerSource: 'Battery',
            endpoints: [
               {ID: 1, profileID: 260, deviceID: 10, inputClusters: [0,9,10,257,64512,1], outputClusters: []},
               {ID: 196, profileID: 260, deviceID: 10, inputClusters: [1], outputClusters: []},
            ]},
        ],
        model: 'ZYA-C4-MOD-S',
        vendor: 'Yale',
        description: 'Control4 module for Yale KeyFree/Keyless lock',
        fromZigbee: [fz.lock, fz.lock_operation_event],
        toZigbee: [tz.lock],
        exposes: [e.lock()]
    };

module.exports = definition;
```

ChipWolf, 2021-11-19:
@danieledwardgeorgehitchcock got it working last night with exactly what you have there strangely enough.

I had to add a flow in Node-Red to listen for /set messages and fire a /get message to MQTT a few seconds later to make sure the state is refreshed, would be nice to drop that eventually.

theperm, 2021-11-19:
You bought the C4 module? Or the Yale DWGW-ZB-DLM-YDM? Is it's connector
compatible with the connexis modules. It looks a different shape, not L
shaped

If you are both using C4 module how did you buy it in the UK?

danieledwardgeorgehitchcock, 2021-11-19:
The Homely module (which is the subject of that issue), looks to be the same as the Conexis module.

See the [Homely website entry here](https://www.homely.no/produkter/kommunikasjonsmodul-yale-doorman/)

I have not bought one of the modules as I am assisting Henrik remotely to develop an external converter on his device.

I am using the C4 module. I was lucky enough to find one on eBay but, they are like hens teeth, and unless you have a C4 system supported by a C4 dealer, you have no chance of procuring one outside of that chain!

ChipWolf, 2021-11-19:
> You bought the C4 module? Or the Yale DWGW-ZB-DLM-YDM? Is it's connector compatible with the connexis modules. It looks a different shape, not L shaped If you are both using C4 module how did you buy it in the UK?

I've got the ZYA-C4-MOD-S, bought it from a dealer on a c4forum thread @theperm

ChipWolf, 2021-11-19:
> The Homely module (which is the subject of that issue), looks to be the same as the Conexis module.

My guess is the one referenced in this thread (#6878) is practically the same too @danieledwardgeorgehitchcock

ChipWolf, 2021-12-24:
Certainly not stale

ChipWolf, 2022-02-01:
@Koenkk not stale

ChipWolf, 2022-03-04:
/destale

ChipWolf, 2022-04-09:
> This is a genuine FR but, it only appears to be me doing the dev on this, and due to my Control4 hub being set up and then stopping working a day after i spent a lot of money getting it commissioned, I am not actively working on the module integration at present. I'd therefore recommend to keep it as stale until a point when the dev work can be picked up by myself / someone else

The examples above seem to work @danieledwardgeorgehitchcock, albeit without battery reporting, maybe we should get that merged in first and then open a secondary issue for enhancing it.

ChipWolf, 2022-05-11:
Not stale

closed 2022-06-19 by github-actions[bot]

danieledwardgeorgehitchcock, 2022-09-01:
I am back to working on the converter for this device (apologies for the delay).

I have been doing some sniffing of the device and I have found that the lock state isn't reported automatically by the module on a state change of the lock. What does change however, is there is a report sent to the ZCL Alarms cluster (which I think is labelled as genAlarms in Z2M). The data is as follows:

```
Alarm Code  |  Cluster ID  |  Comments / Obervations
----------------------------------------------------------
0x16        |  0xFC00      |  Unlocked Manually
0x1B        |  0xFC00      |  Locked Automatically (After Timeout)
0x19        |  0xFC00      |  Unlocked By Module (Zigbee)
0x15        |  0xFC00      |  Locked Manually
```

Unlocking via a tag doesn't seem to fire an event which is frustrating, but I will keep digging about

ChipWolf, 2022-09-01:
> I am back to working on the converter for this device (apologies for the delay).
>
>
>
> I have been doing some sniffing of the device and I have found that the lock state isn't reported automatically by [...]

I've resorted to a magnetic contact on the door to refresh state, we can probably infer tag unlocks if we have a signal for manual/module unlocks.

danieledwardgeorgehitchcock, 2022-09-01:
Agreed - It's a bit of a shame though, as I know that the Z-Wave modules report back the tag ID which with some mapping in Home Assistant, would be quite useful to see who has unlocked the door.

What I forgot to mention above, is that you can manually read the lock state in the dev console by sending a read command to closuresDoorLock with for the attribute of lockState. I am wondering if a read of this attribute could be triggered after the above alarm state change.

As for the battery status, I am not finding any clusters / attributes that report that back. It may well be coming back in the proprietary Control4 payload

danieledwardgeorgehitchcock, 2022-09-01:
One of the biggest hurdles that I am facing, is that I don't know how the lock interface looks in Control4 - it was an expensive ordeal to get my hub set up, only for it to reset back to defaults before I could see anything.

It will be some time before I can afford to get that set back up.

If I could source some screenshots of how the lock looks in the C4 interface - that would help a bunch. Other than that, I am trying to integrate the device blind which takes a lot of time

danieledwardgeorgehitchcock, 2022-09-15:
I have just created a Pull Request for the basic lock / unlock functionality (hopefully it will get approved).

I just wanted to document some things in case others may want to contribute to getting more functionality for this device working in the future (I will also continue to work on it).

I have managed to get the module linked up to my Control4 hub and the Control4 Composer program shows the following attributes and commands for the module:

Attributes
[image]

Commands
[image]

I also found the driver and extracted it with 7-Zip - there is a lot of interesting stuff in there including documentation, OTA firmware's and some potentially interesting XML files. Unfortunately, the driver itself is written in encrypted LUA so, I cannot deconstruct it. The extracted driver can be found [here](https://drive.google.com/drive/folders/1niLhcoZPvOc45POFdjRYjZw7b2XHGF_-?usp=sharing)

I am able to ssh into my Control4 hub and run the command line application `zman` which outputs debug Zigbee logs on the fly which is extremely useful for working out how things work.

On-top of being able to see the debug logs on the hub, I can now also sniff the traffic flowing between the module and the hub therefore negating the actual need to decrypt the LUA.

It seems that depending on which lock model you get, different attributes are reported back from the module - RFID open events don't for instance, seem to be reported back from the module when paired with my Yale Conexis L1 lock so, I can only code what I am seeing - keypad locks for instance may report a different alarm state however, the device handler that I have created should handle that and raise it as a warning which could then be added to the list.

The battery state is my next thing to work on - this is a pain to track down - It is not a case of reading any of the attributes in the `genPower` cluster - I think it may be coming back from the `genAlarms` cluster which many of the other events seem to be emanating from.

If you look at the sniffs captured earlier in this thread, you will see that there is an additional Control4 layer / protocol that sits on top of the standard Zigbee stack. Some reports come back from the device on that layer - These seem to be manufacturer and version information - I am yet to find a way of reading this data in Z2M.

ChipWolf, 2022-12-20:
@danieledwardgeorgehitchcock could you potentially dump the zman logs somewhere for interest? Thanks!

danieledwardgeorgehitchcock, 2023-01-28:
Hi all,

It's been a while since I have posted an update in here but I come baring good news..

Thanks to a lot of help by some amazing people in the Control4 forums, I have managed to successfully decrypt the lua driver file!

Some interesting stuff that I can see just on a quick scrub; lock model identifiers, a couple of new yale specific clusters, and command and attribute identifiers.

Please find it [here](https://drive.google.com/file/d/1rFt38R2TUPKdwicay126iD2RtWqjUBxR/view?usp=sharing)

danieledwardgeorgehitchcock, 2023-01-30:
Been doing some deeper digging this afternoon - it appears that all of the calls specified in the requires of the lua file are all elsewhere in the code so it is a self-contained, self-referencing file - Good news for picking it apart!

I'm going to work on the command and attribute clusters over the coming days. Once that's in Z2M, its a case of writing the fromZigbee and toZigbee converters which is reasonably easy to do

danieledwardgeorgehitchcock, 2023-01-30:
My (untested) addition to `zigbee-herdsman\src\zcl\definition\cluster.ts`:

```javascript
    manuSpecificAssaDoorLock: {
        ID: 0xeacc,
        attributes: {
            autoLockTime: {ID: 0x0012, type: DataType.uint8},
            wrongCodeAttempts: {ID: 0x0013, type: DataType.uint8},
            shutdownTime: {ID: 0x0014, type: DataType.uint8},
            batteryLevel: {ID: 0x0015, type: DataType.uint8},
            insideEscutcheonLED: {ID: 0x0016, type: DataType.uint8},
            volume: {ID: 0x0017, type: DataType.uint8},
            lockMode: {ID: 0x0018, type: DataType.uint8},
            language: {ID: 0x0019, type: DataType.uint8},
            allCodesLockout: {ID: 0x001A, type: DataType.boolean},
            oneTouchLocking: {ID: 0x001B, type: DataType.boolean},
            privacyButtonSetting: {ID: 0x001C, type: DataType.boolean},
            /* enableLogging: {ID: 0x0020, type: DataType.boolean},*/ // marked in C4 driver as not supported
            numberLogRecordsSupported: {ID: 0x0021, type: DataType.uint16},
            numberPinsSupported: {ID: 0x0030, type: DataType.uint8},
            numberScheduleSlotsPerUser: {ID: 0x0040, type: DataType.uint8},
            alarmMask: {ID: 0x0050, type: DataType.uint8},
        },
        commands: {
            getLockStatus: {
                ID: 0x10,
                response: 2,
                parameters: [],
            },
            getBatteryLevel: {
                ID: 0x12,
                parameters: [],
            },
            setRFLockoutTime: {
                ID: 0x13,
                parameters: [],
            },
            /* getLogRecord: {
                ID: 0x20,
                parameters: [],
            },*/ // marked in C4 driver as not supported
            userCodeSet: {
                ID: 0x30,
                parameters: [
                    {name: 'payload', type: DataType.charStr}, // bit pack ("bbb", slot, status, pinLength) .. pin
                ],
            },
            userCodeGet: {
                ID: 0x31,
                parameters: [
                    {name: 'payload', type: DataType.charStr}, // bit pack ("b", slot)
                ],
            },
            userCodeClear: {
                ID: 0x32,
                parameters: [
                    {name: 'payload', type: DataType.charStr}, // bit pack ("b", slot)
                ],
            },
            clearAllUserCodes: {
                ID: 0x33,
                parameters: [],
            },
            setUserCodeStatus: {
                ID: 0x34,
                parameters: [],
            },
            getUserCodeStatus: {
                ID: 0x35,
                parameters: [],
[... 120 lines cut ...]
                parameters: [
                    {name: 'status', type: DataType.uint8},
                ],
            },
            reflashDataRsp: {
                ID: 5,
                parameters: [
                    {name: 'status', type: DataType.uint8},
                ],
            },
            reflashStatusRsp: {
                ID: 6,
                parameters: [
                    {name: 'status', type: DataType.uint8},
                ],
            },
            /* userStatusRsp: {
                ID: 7,
                parameters: [
                    {name: 'status', type: DataType.uint8},
                ],
            },*/ // C4 driver has this response yet there is no command - maybe a non-specific cluster response?
        },
    },
```

t112013, 2023-02-01:
can this be the same module ?
[https://www.elektroimportoren.no/futurehome-zigbee-yale-doorman-modul/94350/Product.html](url)

danieledwardgeorgehitchcock, 2023-02-02:
I tried to find someone who'd be willing to send me one to find out.

Short answer is probably a bit of a yes and no.

Through looking at the PCB design of the Control4 module, it looks to be mass produced. It's therefore highly likely that the PCB's between the various Zigbee modules (there's some Brazillian modules floating about too) will be the same.

There will likely be a difference in the firmware between the modules. For instance, my module seems to report some messages on a special Control4 layer.

All is not lost however, as I believe that the base firmware on the the modules will hopefully be the same, and that it has just been extended to suit the various hardware manufacturers.

It's likely that at least a large chunk of the effort and work that has been done on this module, could likely be re-purposed for the others.

I have added the Doorman cluster as part of my PR so that someone could maybe investigate

danieledwardgeorgehitchcock, 2023-02-02:
For anyone that wants to help out from this point, I have added some nots of the parameter formats in the clusters PR that has now been merged above. The bit pack function datatypes are as follows:

```lua
ZCL.DataTypes.NameToFormat = {
    data8 = "b",
    data16 = "H",
    data32 = "L",
    bool = "b",
    map8 = "b",
    map16 = "H",
    map32 = "L",
    uint8 = "b",
    uint16 = "H",
    uint32 = "L",
    int8 = "b",
    int16 = "h",
    int32 = "l",
    enum8 = "b",
    enum16 = "H",
    single = "f",
    double = "d",
    octstr = "p",
    string = "p",
    UTC = "I"
}
```
The `fromZigbee` equivalents in the driver are the functions that contain Inbound at the start of the title such as `Inbound.Handlers.AssaDoorLock.GetLockStatus`

I will be focusing my efforts on getting the battery attribute working as I think this is a good one to start with, and will likely be a high value for the device driver.

danieledwardgeorgehitchcock, 2024-01-08:
Added some new features in Koenkk/zigbee-herdsman-converters#6713