Koenkk/zigbee2mqtt #4779: bulb firmware doesn't update

gabrielke, 2020-10-26:
## What happened

Bulb firmware didn't update to newer version.

INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"iroasztal_lampa"},"meta":{"device":"iroasztal_lampa","from":{"date_code":"20190308","software_build_id":"2.0.022"},"status":"update_succeeded","to":{"date_code":"20190308","software_build_id":"2.0.022"}},"type":"ota_update"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20190308","software_build_id":"2.0.022"},"id":"iroasztal_lampa","to":{"date_code":"20190308","software_build_id":"2.0.022"}},"status":"ok"}'

## What did you expect to happen

I expected it to update to 2.0.029

## How to reproduce it (minimal and precise)

This happens with this specific bulb everytime.
Now the Conbee device is not more than 1m away from the bulb but still doesn't update.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: Conbee II
Adapter firmware version:

Koenkk, 2020-10-26:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/4559

closed 2020-10-26 by Koenkk

gabrielke, 2020-10-28:
I had the same issue with a tradfri 1810 switch.
I just removed the battery overnight and next day the switch booted with the new firmware.