Koenkk/zigbee2mqtt #4783: Danfoss eTRV thermostat valve movement inconsistent

Zeddy-, 2020-10-26:
I am using Danfoss Ally thermostats, and I thought they were working fine. However now I've realized that changing the setpoint, instantly updated the display, but the valve can sometimes take hours to adjust. 

[image]

Since the goal is to have full control of the Valve position and be able to quickly fully open / close it, i would have to somehow work around the built in PID, which i think is the main culprit here.

What is weird is that yesterday valves weren't moving at all, this morning they were updating instantly, going from fully closed to fully opened, over and over again. Until later in the day, the Valves stopped working again. 

How could this be? I realize this is not a zigbee2mqtt issue, but i imagine this is the best forum on the planet for help with these topics. 

Also does anyone have any idea what these "commands" are, i get the setpoint raise/lower, but what is this Setpoint Command supposed to be?
[image]

## Debug info (Hardware makes no difference)
Zigbee2MQTT version: latest 
Adapter hardware: CC2531 / CC2530 / CC2530 + CC2591 DL22

randomuser389, 2020-10-27:
This is not a problem, the idea about the product is that it has build in PID.
It may have to calibrate, it does this the first week of installation, it needs to find the opening point.
The zigbee setpoint command with first byte (enum8) set to 1 and next 2 bytes set to setpoint (16bit) will be more aggressive then setting occupied heating setpoint.

Zeddy-, 2020-10-28:
> This is not a problem, the idea about the product is that it has build in PID.
> It may have to calibrate, it does this the first week of installation, it needs to find the opening point.
> The zigbee [...]

Thank you very much for clearing this up. I haven't been able to find how to execute commands using zigbee2mqtt, is it similar to reading/writing attributes?

zhapwey, 2020-11-03:
Also interested in how to do this, because the trv looks good but its movements are very very slow.

thanks

closed 2020-12-12 by github-actions[bot]

ma-zal, 2020-12-14:
For this issue I opened topic [Danfoss Ally: Slow actuator response](https://community.home-assistant.io/t/danfoss-ally-slow-actuator-response/254939) in Community HA.

robertalexa, 2021-05-22:
This has now been resolved alongside a significant amount of improvements to this TRV.
Koenkk/zigbee-herdsman-converters#2592
The changes are now available on the dev branch of z2m