Koenkk/zigbee2mqtt #4785: Error while starting zigbee-herdsman, since new slaesh stick,

jbeumer, 2020-10-26:
I installed my new slaesh adapter last week, and it worked for three days, after that it needed restarting zigbee2mqtt for two days to get it online and since then it stopped working. Plugin was uninstalled, installed again, restarted home assistant couple times, tried a powered usb hub my I am running out of options. Hopefully somebody sees some obvious things in the logging below. 

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Zigbee Herdsman debug logging enabled.
[Info] Socat is DISABLED and not started
2020-10-26T18:48:30: PM2 log: Launching in no daemon mode
2020-10-26T18:48:30: PM2 log: App [npm:0] starting in -fork mode-
2020-10-26T18:48:30: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-26 18:48:36: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-26.18-48-36' filename: log.txt
Zigbee2MQTT:debug 2020-10-26 18:48:36: Removing old log directory '/share/zigbee2mqtt/log/2020-10-26.18-36-32'
Zigbee2MQTT:debug 2020-10-26 18:48:37: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-10-26 18:48:38: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-26 18:48:38: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-26 18:48:38: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"}}'
2020-10-26T18:48:38.630Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-10-26T18:48:38.635Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-10-26T18:48:38.643Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-10-26T18:48:38.664Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-10-26T18:48:39.011Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2020-10-26T18:48:39.015Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2020-10-26T18:48:39.146Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-10-26T18:48:39.148Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-10-26T18:48:40.164Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-26T18:48:40.171Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-10-26T18:48:42.010Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2020-10-26T18:48:42.011Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254,6,65,128,0,2,1,2,7,1,192]
2020-10-26T18:48:42.014Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2020-10-26T18:48:42.021Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2020-10-26T18:48:42.026Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-26T18:48:46.177Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-26T18:48:46.180Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-10-26T18:48:52.188Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-26T18:48:52.191Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-10-26T18:48:52.197Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2020-10-26T18:48:52.198Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2020-10-26T18:48:52.199Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2020-10-26T18:48:52.201Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2020-10-26T18:48:52.203Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-26T18:48:52.207Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-10-26T18:48:52.208Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-10-26T18:48:58.211Z zigbee-herdsman:adapter:zStack:adapter Failed to get zStack version, assuming 1.2
2020-10-26T18:48:58.212Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2
2020-10-26T18:48:58.212Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""})
2020-10-26T18:48:58.221Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-10-26T18:48:58.229Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37]
Zigbee2MQTT:error 2020-10-26 18:49:04: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-26 18:49:04: Failed to start zigbee
Zigbee2MQTT:error 2020-10-26 18:49:04: Exiting...
Zigbee2MQTT:error 2020-10-26 18:49:04: Error: SRSP - SYS - osalNvRead after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-26T18_49_04_318Z-debug.log
2020-10-26T18:49:04: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-10-26T18:49:04: PM2 log: App [npm:0] starting in -fork mode-
2020-10-26T18:49:04: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-26 18:49:09: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-26.18-49-09' filename: log.txt
Zigbee2MQTT:debug 2020-10-26 18:49:09: Removing old log directory '/share/zigbee2mqtt/log/2020-10-26.18-36-59'
Zigbee2MQTT:debug 2020-10-26 18:49:11: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-10-26 18:49:11: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-26 18:49:11: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-26 18:49:11: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"}}'
2020-10-26T18:49:11.913Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
[... cut ...]
2020-10-26T18:49:12.428Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-10-26T18:49:12.430Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-10-26T18:49:13.448Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-26T18:49:13.462Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-10-26T18:49:19.472Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-26T18:49:19.475Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]`

jbeumer, 2020-10-27:
Restarting the whole thing a couple times solved the issue again...

ovizii, 2020-10-30:
@jbeumer sorry to jump onto your thread but I have a similar problem. I purchased the exact same zigbee stick you have and am struggling. I also see this error in my zigbee2mqtt home assistent addon:

2020-10-26T18:57:08.429Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'

after that, everything in the logs seem good except I can'T pair a single one of my xiaomi devices. I know they are a bit difficult sometimes so I tried many times.

I am just unsure if above error line means my stick is or is not working? I am about to get an extra non-xiaomi zigbee sensor just to confirm my zigbee stick is actually working.

Please let me know if you have any pointers whatsoever. What does the log look like after thigsn started working on your end, is this line gone?

2020-10-26T18:57:08.429Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'

jbeumer, 2020-10-31:
have to reopen this issue, same error as @ovizii is coming up now. Stick is most off the time not working and have to restart zigbee2mqtt every two hours.

Koenkk, 2020-11-01:
Please provide the herdsman debug logging of this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Ginkodan, 2020-11-01:
Hi , i am using a zzh! CC2652R Stick but might have the same problem:

`[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Zigbee Herdsman debug logging enabled.
[Info] Socat is DISABLED and not started
2020-11-01T12:28:11: PM2 log: Launching in no daemon mode
2020-11-01T12:28:11: PM2 log: App [npm:0] starting in -fork mode-
2020-11-01T12:28:11: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-11-01 12:28:12: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-01.12-28-12' filename: log.txt
Zigbee2MQTT:debug 2020-11-01 12:28:12: Removing old log directory '/share/zigbee2mqtt/log/2020-11-01.12-18-17'
Zigbee2MQTT:debug 2020-11-01 12:28:13: Can't load state from file /share/zigbee2mqtt/state.json (doesn't exist)
Zigbee2MQTT:info  2020-11-01 12:28:13: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-11-01 12:28:13: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-11-01 12:28:13: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6758},"serialPort":{"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"}}'
2020-11-01T11:28:13.149Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-11-01T11:28:13.151Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6758,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-11-01T11:28:13.152Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-11-01T11:28:13.166Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-11-01T11:28:13.638Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-11-01T11:28:13.639Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-11-01T11:28:14.641Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-01T11:28:14.643Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-01T11:28:14.649Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2020-11-01T11:28:14.650Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2020-11-01T11:28:14.651Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2020-11-01T11:28:14.652Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2020-11-01T11:28:14.653Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-01T11:28:14.653Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-11-01T11:28:14.654Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-11-01T11:28:14.661Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-11-01T11:28:14.662Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-11-01T11:28:14.663Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,1,2,7,1,221,61,52,1,0,255,255,255,255] - 191
2020-11-01T11:28:14.664Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200925}
2020-11-01T11:28:14.664Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-01T11:28:14.664Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16`

Same problem with:
2020-11-01T11:28:13.149Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'

jbeumer, 2020-11-03:
And the problem is back:

`Zigbee2MQTT:debug 2020-11-03 20:39:16: Removing old log directory '/share/zigbee2mqtt/log/2020-11-03.20-34-30'
Zigbee2MQTT:info  2020-11-03 20:39:16: Starting Zigbee2MQTT version 1.16.0 (commit #1.16.0)
Zigbee2MQTT:info  2020-11-03 20:39:16: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-11-03 20:39:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"}}'
2020-11-03T20:39:17.238Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-11-03T20:39:17.243Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-11-03T20:39:17.254Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-11-03T20:39:17.273Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-11-03T20:39:17.752Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-11-03T20:39:17.754Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-11-03T20:39:18.770Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:18.783Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-03T20:39:24.794Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:24.796Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-03T20:39:30.804Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-03T20:39:30.807Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2020-11-03 20:39:36: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-03 20:39:36: Failed to start zigbee
Zigbee2MQTT:error 2020-11-03 20:39:36: Exiting...
Zigbee2MQTT:error 2020-11-03 20:39:36: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-03T20_39_36_893Z-debug.log
2020-11-03T20:39:37: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-03T20:39:37: PM2 log: App [npm:0] starting in -fork mode-
2020-11-03T20:39:38: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.0 start /zigbee2mqtt-1.16.0`

jbeumer, 2020-11-07:
It crashes every day:

```
Zigbee2MQTT:debug 2020-11-07 18:40:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"}}'
2020-11-07T18:40:56.101Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-11-07T18:40:56.107Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-11-07T18:40:56.118Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_02_79-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-11-07T18:40:56.137Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-11-07T18:40:56.618Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-11-07T18:40:56.620Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-11-07T18:40:57.630Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-07T18:40:57.642Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-07T18:41:03.653Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-07T18:41:03.656Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-07T18:41:03.665Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2020-11-07T18:41:03.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2020-11-07T18:41:03.672Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2020-11-07T18:41:03.680Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2020-11-07T18:41:03.685Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-07T18:41:03.686Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-11-07T18:41:03.687Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-11-07T18:41:09.689Z zigbee-herdsman:adapter:zStack:adapter Failed to get zStack version, assuming 1.2
2020-11-07T18:41:09.690Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2
2020-11-07T18:41:09.691Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""})
2020-11-07T18:41:09.699Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-11-07T18:41:09.705Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37]
Zigbee2MQTT:error 2020-11-07 18:41:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-07 18:41:15: Failed to start zigbee
Zigbee2MQTT:error 2020-11-07 18:41:15: Exiting...
Zigbee2MQTT:error 2020-11-07 18:41:15: Error: SRSP - SYS - osalNvRead after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-07T18_41_15_790Z-debug.log
2020-11-07T18:41:16: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-07T18:41:16: PM2 log: App [npm:0] starting in -fork mode-
2020-11-07T18:41:16: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
```

Koenkk, 2020-11-07:
Make sure to check the following first: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start, especially:

- https://www.zigbee2mqtt.io/information/FAQ.html#for-openhab-users-disable-zwave-binding
- https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed
- https://www.zigbee2mqtt.io/information/FAQ.html#hciuart-is-running

jbeumer, 2020-11-07:
I guess the error is related to the new stick, I did have some issues with my cc2531 stick but the struggle got worse when I started using the new slaesh stick.

Ginkodan, 2020-11-08:
I moved my zigbee2mqtt docker from Home Assistant directly to unRaid and now it works. Probably a problem with USB and Home Assistant.

jbeumer, 2020-11-08:
> I moved my zigbee2mqtt docker from Home Assistant directly to unRaid and now it works.

I have no idea how to fix that in home assistant, could you describe on how to do that?

golddragon007, 2020-11-12:
@Koenkk After I had a fail like in #4398 I had this error message like the OP for me it recovered somehow by itself after a few failed attempt, here is the log: 
[network_fail-after-restart.zip](https://github.com/Koenkk/zigbee2mqtt/files/5532569/network_fail-after-restart.zip)

eberlemartin, 2020-11-17:
Same error here since today:
`Zigbee2MQTT:error 2020-11-17 21:45:43: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)`

Im running zigbee2mqtt with HomeAssistant in docker on a RPi4. Today i paired some now ikea bulbs, then suddenly my switches stopped working and after a reboot of the host this error appears when i try to start zigbee2mqtt plugin. I tried all the steps in the readme regarding usb ports etc. but no luck.

Please help :)

closed 2020-12-25 by github-actions[bot]

krema, 2021-03-24:
@eberlemartin could you find a solution? That's exactly what happened to me. 2 new Ikea lamps and bang "Error: SRSP - SYS - Ping after 6000ms" error.

eberlemartin, 2021-03-24:
After a long search i ended up actually buying the stick from slaesh. Apparently pairing multiple devices at once can overwhelm the cc2531 and cause the programming to break, then you have to flash it again. Make sure you exhausted all other tipps first like using the by-id address for your usb port etc.

jla1710, 2021-09-12:
I'm getting the error:

[22:04:00] INFO: Zigbee Herdsman debug logging enabled
[22:04:00] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.21.1-dev start
> node index.js
Zigbee2MQTT:info  2021-09-12 22:04:06: Logging to console and directory: '/config/zigbee2mqtt/log/2021-09-12.22-04-04' filename: log.txt
Zigbee2MQTT:info  2021-09-12 22:04:06: Starting Zigbee2MQTT version 1.21.1-dev (commit #8c0827a)
Zigbee2MQTT:info  2021-09-12 22:04:06: Starting zigbee-herdsman (0.13.143)
2021-09-12T19:04:06.756Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2021-09-12T19:04:06.761Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6666,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2021-09-12T19:04:06.763Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2021-09-12T19:04:06.782Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2021-09-12T19:04:06.797Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2021-09-12T19:04:06.799Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2021-09-12T19:04:07.107Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2021-09-12T19:04:07.108Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0]
2021-09-12T19:04:07.251Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2021-09-12T19:04:07.252Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2021-09-12T19:04:08.158Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2021-09-12T19:04:08.159Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0,254,6,65,128,0,2,1,2,7,1,192]
2021-09-12T19:04:08.162Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2021-09-12T19:04:08.168Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2021-09-12T19:04:08.170Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-12T19:04:08.260Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-09-12T19:04:08.264Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-09-12T19:04:08.269Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2021-09-12T19:04:08.269Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2021-09-12T19:04:08.269Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2021-09-12T19:04:08.270Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2021-09-12T19:04:08.272Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-12T19:04:08.272Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2021-09-12T19:04:08.273Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2021-09-12T19:04:08.277Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,66,62,52,1,0,39]
2021-09-12T19:04:08.277Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,66,62,52,1,0,39]
2021-09-12T19:04:08.277Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,66,62,52,1,0] - 39
2021-09-12T19:04:08.278Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20201026}
2021-09-12T19:04:08.279Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-12T19:04:08.279Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2021-09-12T19:04:08.279Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20201026})
2021-09-12T19:04:08.281Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2021-09-12T19:04:08.283Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {}
2021-09-12T19:04:08.283Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37]
2021-09-12T19:04:08.287Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,8,97,4,106,58,204,33,0,75,18,0,137]
2021-09-12T19:04:08.287Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,106,58,204,33,0,75,18,0,137]
2021-09-12T19:04:08.287Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 8 - 3 - 1 - 4 - [106,58,204,33,0,75,18,0] - 137
2021-09-12T19:04:08.288Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - getExtAddr - {"extaddress":"0x00124b0021cc3a6a"}
2021-09-12T19:04:08.288Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-12T19:04:08.291Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":130}
2021-09-12T19:04:08.293Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,130,0,178]
2021-09-12T19:04:08.299Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,0,0,112]
2021-09-12T19:04:08.299Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,0,0,112]
[... cut ...]
Zigbee2MQTT:error 2021-09-12 22:04:08: TypeError: Cannot read property 'length' of null
    at AdapterNvMemory.init (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:28:42)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:60:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.js:113:27)
    at start (/app/index.js:95:5)