Koenkk/zigbee2mqtt #4789: TRADFRI SHORTCUT button (E1812) two events with different groupID

joshua-schnabel, 2020-10-27:
## What happened

I wanted to add the new TRADFRI SHORTCUT button (E1812) to my zigbee2mqtt instance. Therefore I copied the configuration of E1743 and adjusted it. The pairing works without problems. Also sending events by pressing the button works.
```
    {
        zigbeeModel: ['TRADFRI SHORTCUT Button'],
        model: 'E1812',
        vendor: 'IKEA',
        description: 'IKEA TRADFRI Shortcut Button',
        supports: 'on, off, brightness up/down/stop',
        fromZigbee: [
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery
        ],
        toZigbee: [],
        ota: ota.tradfri,
        meta: {configureKey: 1, disableActionGroup: true, battery: {dontDividePercentage: true}},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.bind('genOnOff', defaultBindGroup);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

The used configuration should set the groupID to 901. With the first button press (genOnOff) this is done correctly. But from the second press on two events are received. One with groupID 901 and immediately afterwards with groupID 0.

**First press:**

https://pastebin.com/PbTMTvLz Line 908

**Second press:**

https://pastebin.com/PbTMTvLz Line 931 (second message: 943)

## What did you expect to happen

I am expecting one event per press with groupID 901 and no second event.

## How to reproduce it (minimal and precise)

Add device to devices.js and pair E1812. Press E1812 two times.

## Debug info
Zigbee2MQTT version: dev
Adapter hardware: CC2531 
Adapter firmware version: 20190619

Koenkk, 2020-10-28:
In the `configure()`, can you try to unbind genOnOff from groupID 0? Can be done by adding: `await endpoint.unbind('genOnOff',0);`. Make sure to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure afterwards and check the log if this was OK (you might need to wakeup the device right before sending the command by clicking a button on it).

joshua-schnabel, 2020-10-28:
Thank you very much for your feedback. Unfortunately the behaviour is the same. First trigger results in one event, from the second trigger on there are two events with different groupIDs.

First press: 2020-10-28 18:28:10
Second Press: 2020-10-28 18:28:12

Used Config:

```
    {
        zigbeeModel: ['TRADFRI SHORTCUT Button'],
        model: 'E1812',
        vendor: 'IKEA',
        description: 'IKEA TRADFRI Shortcut Button',
        supports: 'on, off, brightness up/down/stop',
        fromZigbee: [
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery
        ],
        toZigbee: [],
        ota: ota.tradfri,
        meta: {configureKey: 1, disableActionGroup: true, battery: {dontDividePercentage: true}},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
			await endpoint.unbind('genOnOff',0);
            await endpoint.bind('genOnOff', defaultBindGroup);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

Logfiles: 

```
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":901,"clusterid":6,"srcaddr":37646,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":157,"securityuse":0,"timestamp":319020,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[1,2,1]}} +29s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":37646,"endpoint":1,"linkquality":157,"groupID":901}' +31s
Zigbee2MQTT:debug 2020-10-28 18:28:10: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-10-28 18:28:10: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":21,"linkquality":157}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0x680ae2fffe2f3d69/1 6(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +31s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x680ae2fffe2f3d69:37646/1 (0,0,1) +31s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":37646,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":21,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,2,11,1,0]}} +31s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,14,147,1,1,6,0,21,0,30,5,24,2,11,1,0,175] +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,133,3,6,0,14,147,1,1,0,162,0,250,246,4,0,0,3,1,3,1,14,147,11,243] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,133,3,6,0,14,147,1,1,0,162,0,250,246,4,0,0,3,1,3,1,14,147,11,243] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [133,3,6,0,14,147,1,1,0,162,0,250,246,4,0,0,3,1,3,1,14,147,11] - 243 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":901,"clusterid":6,"srcaddr":37646,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":162,"securityuse":0,"timestamp":325370,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[1,3,1]}} +2s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":37646,"endpoint":1,"linkquality":162,"groupID":901}' +2s
Zigbee2MQTT:debug 2020-10-28 18:28:12: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-10-28 18:28:12: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":21,"linkquality":162}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0x680ae2fffe2f3d69/1 6(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,0,0,6,0,14,147,1,1,0,162,0,17,247,4,0,0,3,1,3,1,14,147,29,137] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,14,147,1,1,0,162,0,17,247,4,0,0,3,1,3,1,14,147,29,137] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,14,147,1,1,0,162,0,17,247,4,0,0,3,1,3,1,14,147,29] - 137 +0ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,23,209] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,23,209] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,23] - 209 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":23} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
```

Koenkk, 2020-10-29:
- I find it really strange that it only produces one event for the first click
- Did you check if the reconfigure was OK? (did you see `Succesfully configured...` after https://github.com/Koenkk/zigbee2mqtt/issues/4789#issuecomment-718047348)
- If we cannot unbind it from group 0 would be really problematic since some devices are by default in this group.

joshua-schnabel, 2020-10-30:
Hello, I have once again set up everything anew to eliminate all influences. I also changed the key and the channel of the network.

Yes, I got the "Succesfully configured...". See log files below. 

Attached you find the log file which contains the complete process. Only the very first click after adding the button is not sent twice. After a reconfiguration the first click is sent twice.

> Zigbee2MQTT:debug 2020-10-30 10:55:26: Can't load state from file D:\Neuer Ordner\zigbee2mqtt\data\state.json (doesn't exist)
> Zigbee2MQTT:info  2020-10-30 10:55:26: Logging to console and directory: 'D:\Neuer Ordner\zigbee2mqtt\data\log\2020-10-30.10-55-21' filename: log.txt
> Zigbee2MQTT:info  2020-10-30 10:55:26: Starting Zigbee2MQTT version 1.15.0-dev (commit #21fcfb0)
> Zigbee2MQTT:info  2020-10-30 10:55:26: Starting zigbee-herdsman...
> Zigbee2MQTT:debug 2020-10-30 10:55:26: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"D:\\Neuer Ordner\\zigbee2mqtt\\data\\coordinator_backup.json","databaseBackupPath":"D:\\Neuer Ordner\\zigbee2mqtt\\data\\database.db.backup","databasePath":"D:\\Neuer Ordner\\zigbee2mqtt\\data\\database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"\\\\.\\COM5"}}'
> Zigbee2MQTT:info  2020-10-30 10:55:34: zigbee-herdsman started
> Zigbee2MQTT:info  2020-10-30 10:55:34: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
> Zigbee2MQTT:debug 2020-10-30 10:55:34: Zigbee network parameters: {"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6754}
> Zigbee2MQTT:info  2020-10-30 10:55:34: Currently 0 devices are joined:
> Zigbee2MQTT:warn  2020-10-30 10:55:34: `permit_join` set to  `true` in configuration.yaml.
> Zigbee2MQTT:warn  2020-10-30 10:55:34: Allowing new devices to join.
> Zigbee2MQTT:warn  2020-10-30 10:55:34: Set `permit_join` to `false` once you joined all devices.
> Zigbee2MQTT:info  2020-10-30 10:55:34: Zigbee: allowing new devices to join.
> Zigbee2MQTT:info  2020-10-30 10:55:34: Connecting to MQTT server at mqtt://atlantis.fritz.box
> Zigbee2MQTT:info  2020-10-30 10:55:34: Connected to MQTT server
> Zigbee2MQTT:info  2020-10-30 10:55:34: MQTT publish: topic 'zigbee2mqttDEV/bridge/state', payload 'online'
> Zigbee2MQTT:info  2020-10-30 10:55:34: MQTT publish: topic 'zigbee2mqttDEV/bridge/config', payload '{"commit":"21fcfb0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.15.0-dev"}'
> Zigbee2MQTT:info  2020-10-30 10:56:03: Device '0x680ae2fffe2f3d69' joined
> Zigbee2MQTT:info  2020-10-30 10:56:03: Starting interview of '0x680ae2fffe2f3d69'
> Zigbee2MQTT:info  2020-10-30 10:56:03: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe2f3d69"},"type":"device_connected"}'
> Zigbee2MQTT:info  2020-10-30 10:56:03: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffe2f3d69"},"type":"pairing"}'
> Zigbee2MQTT:debug 2020-10-30 10:56:05: Device '0x680ae2fffe2f3d69' announced itself
> Zigbee2MQTT:info  2020-10-30 10:56:05: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x680ae2fffe2f3d69"},"type":"device_announced"}'
> Zigbee2MQTT:debug 2020-10-30 10:56:26: Received Zigbee message from '0x680ae2fffe2f3d69', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI SHORTCUT Button"}' from endpoint 1 with groupID 0
> Zigbee2MQTT:debug 2020-10-30 10:56:26: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI SHORTCUT Button"}'
> Zigbee2MQTT:debug 2020-10-30 10:56:28: Received Zigbee message from '0x680ae2fffe2f3d69', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
> Zigbee2MQTT:debug 2020-10-30 10:56:28: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"IKEA of Sweden"}'
> Zigbee2MQTT:debug 2020-10-30 10:56:30: Received Zigbee message from '0x680ae2fffe2f3d69', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
> Zigbee2MQTT:debug 2020-10-30 10:56:30: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
> Zigbee2MQTT:debug 2020-10-30 10:56:32: Received Zigbee message from '0x680ae2fffe2f3d69', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
> Zigbee2MQTT:debug 2020-10-30 10:56:32: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
> Zigbee2MQTT:debug 2020-10-30 10:56:34: Received Zigbee message from '0x680ae2fffe2f3d69', type 'readResponse', cluster 'genBasic', data '{"appVersion":33}' from endpoint 1 with groupID 0
> Zigbee2MQTT:debug 2020-10-30 10:56:34: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":33}'
[... cut ...]
> Zigbee2MQTT:debug 2020-10-30 11:02:01: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
> Zigbee2MQTT:info  2020-10-30 11:02:01: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","linkquality":147}'
> Zigbee2MQTT:debug 2020-10-30 11:02:01: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
> Zigbee2MQTT:info  2020-10-30 11:02:01: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","linkquality":147}'

Koenkk, 2020-10-30:
Seems the device does not allow the unbinding, this should be documented properly after this device has been integrated. I think the only solution would be to skip the binding to `defaultBindGroup`, leave out:

```
	    await endpoint.unbind('genOnOff',0);
            await endpoint.bind('genOnOff', defaultBindGroup);
```

joshua-schnabel, 2020-10-31:
I have continued research on the problem. Thereby I have found further unusual behaviors.  
When I use the following configuration, everything works as expected. One message from 901.
```
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
	    await endpoint.bind('genOnOff', defaultBindGroup);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
```
When I insert the genPowerCfg line, the behavior is again two messages, one of 0 one of 901.

```
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
	    await endpoint.bind('genOnOff', defaultBindGroup);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
```
So I assume that this line produces the unwanted behaviour. I have understood that genPowerCfg is responsible for the battery indicator. Will I get a battery indicator if I leave out the genPowerCfg line? 

Now it becomes really strange. When I add the button to a group via MQTT I get the following log entries. 
```
Zigbee2MQTT:info  2020-10-31 02:07:44: Successfully bound cluster 'genOnOff' from '0x680ae2fffe2f3d69' to 'group_2'
Zigbee2MQTT:info  2020-10-31 02:07:44: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"0x680ae2fffe2f3d69","to":"group_2"},"type":"device_bind"}'
Zigbee2MQTT:debug 2020-10-31 02:07:44: binding cluster 'genLevelCtrl' from '0x680ae2fffe2f3d69' to 'group_2'
Zigbee2MQTT:info  2020-10-31 02:07:48: Successfully bound cluster 'genLevelCtrl' from '0x680ae2fffe2f3d69' to 'group_2'
Zigbee2MQTT:info  2020-10-31 02:07:48: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"0x680ae2fffe2f3d69","to":"group_2"},"type":"device_bind"}'
Zigbee2MQTT:debug 2020-10-31 02:07:48: binding cluster 'closuresWindowCovering' from '0x680ae2fffe2f3d69' to 'group_2'
Zigbee2MQTT:info  2020-10-31 02:07:50: Successfully bound cluster 'closuresWindowCovering' from '0x680ae2fffe2f3d69' to 'group_2'
Zigbee2MQTT:info  2020-10-31 02:07:50: MQTT publish: topic 'zigbee2mqttDEV/bridge/log', payload '{"message":{"cluster":"closuresWindowCovering","from":"0x680ae2fffe2f3d69","to":"group_2"},"type":"device_bind"}'
```
Therefore I assume that the device offers the following clusters: genLevelCtrl, closuresWindowCovering, genOnOff. 
Is this correct?
If I press the button now, I get 5 messages. One with 0, one with 901 and three with the group ID 444. 
```
Zigbee2MQTT:debug 2020-10-31 02:08:03: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-10-31 02:08:03: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":131}'
Zigbee2MQTT:debug 2020-10-31 02:08:03: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-31 02:08:03: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":131}'
Zigbee2MQTT:debug 2020-10-31 02:08:03: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 444
Zigbee2MQTT:info  2020-10-31 02:08:03: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":131}'
Zigbee2MQTT:debug 2020-10-31 02:08:03: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 444
Zigbee2MQTT:info  2020-10-31 02:08:03: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":131}'
Zigbee2MQTT:debug 2020-10-31 02:08:03: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 444
Zigbee2MQTT:info  2020-10-31 02:08:03: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":131}'
Zigbee2MQTT:debug 2020-10-31 02:09:07: Received MQTT message on 'zigbee2mqttDEV/bridge/bind/0x680ae2fffe2f3d69' with data 'group_1'
```
Wenn ich den Button in eine zweite Gruppe packe, bekomme ich 8 Nachrichten. Eine Nachricht mit 0, eine mit 901, drei mit gruppe 1 und drei mit gruppe 2. 
```
Zigbee2MQTT:debug 2020-10-31 02:09:28: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-10-31 02:09:28: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":147}'
Zigbee2MQTT:debug 2020-10-31 02:09:28: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-31 02:09:28: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":147}'
Zigbee2MQTT:debug 2020-10-31 02:09:28: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 444
[... cut ...]
Is this behavior due to the device or to zigbee2mqtt? I would be able to live with the first variant if I got a battery indicator. I don't need to add the button to any group.

Thanks in advance for analyzing the problem. I will buy another button on monday to exclude a failure in the device. 
I think it might be a mistake that the button is already in the shelfs. It is not listed anywhere on IKEA, even on its own OTA page it is not listed. Maybe it was a mistake of IKEA to sell it already. Currently it looks like it is not mature.

Koenkk, 2020-10-31:
This device shows indeed very strange beahviour. What happens when you do `await bind(endpoint, 901, ['genPowerCfg']);`, do you get a single message per click and battery reports within 24 hours?

joshua-schnabel, 2020-11-01:
I'm on it.

joshua-schnabel, 2020-11-01:
```
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
	    await endpoint.bind('genOnOff', defaultBindGroup);
            await bind(endpoint, 901, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
```

The upper configuration results in double messages but the battery indicator is transmitted. Theoretically this would be sufficient for me, but I still wonder why the device behaves like this. 

```
Zigbee2MQTT:debug 2020-11-01 15:28:59: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:29:02: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","linkquality":110}'
Zigbee2MQTT:debug 2020-11-01 15:29:02: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:29:02: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","linkquality":110}'
Zigbee2MQTT:debug 2020-11-01 15:32:45: Saving state to file D:\Neuer Ordner\zigbee2mqtt\data\state.json
Zigbee2MQTT:debug 2020-11-01 15:37:45: Saving state to file D:\Neuer Ordner\zigbee2mqtt\data\state.json
Zigbee2MQTT:debug 2020-11-01 15:41:21: Received Zigbee message from '0x680ae2fffe2f3d69', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":34}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:41:21: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"battery":34,"linkquality":120}'
Zigbee2MQTT:debug 2020-11-01 15:42:45: Saving state to file D:\Neuer Ordner\zigbee2mqtt\data\state.json
Zigbee2MQTT:debug 2020-11-01 15:44:29: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:44:29: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":115}'
Zigbee2MQTT:debug 2020-11-01 15:44:29: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:44:29: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":115}'
Zigbee2MQTT:debug 2020-11-01 15:44:32: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:44:32: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":128}'
Zigbee2MQTT:debug 2020-11-01 15:44:32: Received Zigbee message from '0x680ae2fffe2f3d69', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Zigbee2MQTT:info  2020-11-01 15:44:32: MQTT publish: topic 'zigbee2mqttDEV/0x680ae2fffe2f3d69', payload '{"action":"on","battery":34,"linkquality":128}'
```

Koenkk, 2020-11-02:
Great, could you submit a PR? One small thing, please change `await bind(endpoint, 901, ['genPowerCfg']);`  to `await bind(endpoint, defaultBindGroup, ['genPowerCfg']);`

joshua-schnabel, 2020-11-02:
I'm on it.

joshua-schnabel, 2020-11-09:
https://github.com/Koenkk/zigbee2mqtt.io/pull/448
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1739

Here are the requests. I have played with the devices again and with the provided configuration I get a correct behaviour when clicking. When adding to groups there is still the problem that the device sends three messages per click. This can be solved theoretically by a "debounce". 

If everything is fine, the issue can be closed.

MattWestb, 2020-11-09:
The problem is that normally the group and the binding made in the device then touchlink is adding clusters and groups from its inisiator. IKEA have not implanting function for doing the same then doing one classical joining and its making problems like with the 5 button remotes scene commands is being sended to the groupe 0xff09 and cant being changed.

closed 2020-11-09 by Koenkk