Koenkk/zigbee2mqtt #4790: Tradfri not pairing

jonasled, 2020-10-27:
## What happened
I tried pairing 2 new tradfri bulbs, with my zigbee2mqtt using a CC2531 Stick. The problem is, it never shows up in the log (does not start pairing). I tried already a different PC and a different CC2531 Stick and also reflashed one. Pairing with the original tradfri bridge works flawlessly (of course I've reseted the bulb afterwards).

## What did you expect to happen
Device pairs with the bridge.

## How to reproduce it (minimal and precise)
Try to pair a E14 tradfri bulb with the newest firmware.

## Debug info
Zigbee2MQTT version: 1.15.0 (hass.io)
Adapter hardware: CC2531
Adapter firmware version:{"meta": "maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}

io53, 2020-10-27:
Does the bulb flash when you have reset it? I've had some trouble resetting trådfri e14, turned out I was doing the 6 on/off too fast (a problem I've not had with gu10 nor e27 bulbs).

jonasled, 2020-10-27:
No, they don't flash, but first I tried pairing them to the homeassistant and afterwards paired them successfully with the tradfri bridge, so the were resetted I think.

jonasled, 2020-10-27:
Hmm, interesting. I tried the bulb I thought is dead with the tradfri GW and it works there also, but pairing with the homeassistant doesn't work.

jonasled, 2020-10-27:
Ok, some progress. After a lot of tries I've got them reset again. Now they blink forever, but don't pair.

jonasled, 2020-10-27:
I get them slowly to repair after resetting everything to factory defaults. The reset just needs some very hard timing, GG ikea.

closed 2020-10-27 by jonasled