Koenkk/zigbee2mqtt #4791: Has installation fail due to node version 14?

gald33, 2020-10-27:
## What happened
did not see "node-pre-gyp info ok" after last install command "npm ci". When starting with "npm start" I get "Error while starting zigbee-herdsman".

## What did you expect to happen
for the program to run successfuly

## How to reproduce it (minimal and precise)
type "npm start" in /opt/zigbee2mqtt

## Debug info
Zigbee2MQTT version:
zigbee2mqtt@1.15.0

Adapter hardware: CC2531
usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001938766C-if00 -> ../../ttyACM0

Adapter firmware version:
I used the basic recommended version 6 months ago. how to check?

more info:
- I've used these cc2531 and computer last week. I formatted to ubuntu 20.04. now it doesn't work.
- I've added myself to the dialout group and chown the device just in case.
tried with mosquitto broker and mosquitto broker addon in home assistant:
`$ mosquitto -v
[44562.596398]~DLT~1931584~INFO     ~FIFO /tmp/dlt cannot be opened. Retrying later...
1603794558: mosquitto version 1.6.9 starting
1603794558: Using default config.
1603794558: Opening ipv4 listen socket on port 1883.
1603794558: Error: Address already in use
[44572.611527]~DLT~1931584~WARNING  ~Lost log messages in user buffer when exiting: 6
`
error output for npm start:
$ npm start

```
> zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-10-27 12:27:39: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-27.12-27-39' filename: log.txt
Zigbee2MQTT:info  2020-10-27 12:27:39: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
Zigbee2MQTT:info  2020-10-27 12:27:39: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-10-27 12:27:39: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-27 12:27:39: Failed to start zigbee
Zigbee2MQTT:error 2020-10-27 12:27:39: Exiting...
Zigbee2MQTT:error 2020-10-27 12:27:39: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/gal/.npm/_logs/2020-10-27T10_27_39_973Z-debug.log

```
- tried on another pc with the same installation and several USB ports.
- I noticed that the light on the cc2531 turns off after about 10 sec.

has the installation failed or is it something else?

gald33, 2020-10-27:
that's not the problem, installed node.js 12 and everything is the same.
what could be the problem?

MORD0R, 2020-10-28:
Error: Error: Permission denied, cannot open /dev/ttyACM0

1. Determine location of CC2531 USB sniffer and checking user permissions
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

gald33, 2020-10-28:
> Error: Error: Permission denied, cannot open /dev/ttyACM0
> 
>     1. Determine location of CC2531 USB sniffer and checking user permissions
>        https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
location:
```
Both seems fine I think, take a look below:

$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Oct 27 23:24 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001938766C-if00 -> ../../ttyACM0
```
permissions dir:
```
$ ls ../ -l
total 12
drwx--x--x  4 root root 4096 Oct 26 11:36 containerd
drwxr-xr-x  3 root root 4096 Oct 26 11:21 google
drwxrwxrwx 12 gal  gal  4096 Oct 27 23:43 zigbee2mqtt
```

permissions device:
```
$ test -w /dev/ttyACM0 && echo "write permission on" || echo "write permission off"
write permission on
```

here is the error with debug level:
```
$ npm start

> zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-10-28 09:52:17: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-28.09-52-17' filename: log.txt
Zigbee2MQTT:debug 2020-10-28 09:52:17: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-10-28 09:52:17: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
Zigbee2MQTT:info  2020-10-28 09:52:17: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-28 09:52:17: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2020-10-28 09:52:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-28 09:52:58: Failed to start zigbee
Zigbee2MQTT:error 2020-10-28 09:52:58: Exiting...
Zigbee2MQTT:error 2020-10-28 09:52:58: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
```

gald33, 2020-11-23:
Thank you for all the answers. Solved. It turns out that there was an issue with the computer itself, and I had to replace the motherboard. It somehow effected the CC2531 and made it unusable, I think something might have been wrong with the computer voltage which could explain a bunch of problems I has in the following day until it the computer finally crashed. Anyway, I re-flashed the CC2531 and then it got recognized again.

closed 2020-11-23 by gald33