Koenkk/zigbee2mqtt #4792: TZE200_aqnazj70 device support

akinomeroglu, 2020-10-27:
Hi all,

## Information about the device + link

This is device link https://a.aliexpress.com/_AdQdrb (page is in Turkish)

This is device info from log

[],"friendlyName":"0xec1bbdfffef643b0","ieeeAddr":"0xec1bbdfffef643b0","lastSeen":1603748148611,"manufacturerName":"_TZE200_aqnazj70","modelID":"TS0601","networkAddress":17599,"type":"Router"}]}"}

## data/database.db entry of the device

This is database.db entry

{"id":3,"type":"Router","ieeeAddr":"0xec1bbdfffef643b0","nwkAddr":17599,"manufId":4098,"manufName":"_TZE200_aqnazj70","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_aqnazj70","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1603797043769}

FarukGN, 2020-10-30:
@Koenkk Dude where are you

Koenkk, 2020-10-30:
This device is already supported in the latest dev branch: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c638c6af586f862802ef629fb05677d438cf5bf7/devices.js#L1402  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-30 by Koenkk

FarukGN, 2020-10-31:
> This device is already supported in the latest dev branch: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c638c6af586f862802ef629fb05677d438cf5bf7/devices.js#L1402 (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Thank you so much. work with edge without problem. thanks again @Koenkk 

And one more thanks for @akinomeroglu  thanks for your help