Koenkk/zigbee2mqtt #4794: OTA Update fails with ECONNREFUSED

kianusch, 2020-10-27:
When trying to update my IKEA Bulb via the current version of zigbee2mqtt I get an error:

{
  "message": "Update of 'AZ.E27.01' failed (connect ECONNREFUSED 13.225.25.117:80)",
  "meta": {
    "device": "AZ.E27.01",
    "status": "update_failed"
  },
  "type": "ota_update"
}

... could it be, that the IKEA Firmware Site (13.225.25.117) is down?

lorek123, 2020-10-27:
http://fw.ota.homesmart.ikea.net/feed/version_info.json looks fine here, maybe there is something that preventing zigbee2mqtt from connection to the server?

kianusch, 2020-10-27:
What happens if you connect to 13.225.25.117:80 from your browser?

I get a "403 ERROR".

kianusch, 2020-10-27:
Okay - seemed to be a problem with my docker host.  After rebooting everything works fine now.

closed 2020-10-27 by kianusch