Koenkk/zigbee2mqtt #4797: Xiaomi Aqara water leak sensors (SJCGQ11LM) will not stay on the network

alexcam1901, 2020-10-28:
# Bug Report
I have several Xiaomi Aqara water leak sensors (SJCGQ11LM). They will pair and join quickly, but do not continue to periodically check in, and seem to leave the network after a short time. I've had one check back in after an hour, but nothing past that time. 

They will report a water properly when joined. They will rejoin the network when I reset them. They are the only sensors on my network of 40+ sensors with this problem.

## What happened
The sensors do not periodically report in to the controller.

## What did you expect to happen
I expect them to report in at some interval after they join the network, even if they do not detect a water leak.

## How to reproduce it (minimal and precise)
Pair the sensors with the controller and test with water to see they are communicating.  Wait for ~20 - 60 minutes and test again, they will not respond.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: zzh! CC2652R Multiprotocol RF Stick
Adapter firmware version: CC26X2R1_20200925

Zigbee-herdsman debug logging:
https://pastebin.com/GFUjUW6F

alexcam1901, 2020-11-02:
On further testing, I can get these sensors to join the network after many, many power-cycle-reset-remove-rejoin cycles. I think this bug report is that there is no error messaging indicating whether or not the join was successful or not. With other batter-powered sensors, there is a failure message when they do not join, fail the interview, etc..

McGiverGim, 2020-11-06:
If it helps, I had a similar issue but not the same. I have the same sensor, and in the `database.db` file it appears as `"interviewCompleted":false`, but it works perfectly. It seems to report (at least changes in battery and lqi) and it appears in the Map with good lqi.

I have rejoined it and now it appears as `true` so I can't do more tests. If it returns to `false` I will check if it reports the water leak or not.

alexcam1901, 2020-11-06:
That's interesting.  I just looked at the entries for two different (of the same model) sensors in my database.db, and they look almost identical to each other.  One is happily reporting away every 60 minutes, and one is not, both have `interviewCompleted` as `true`. The only thing that looks out of place is the `nwkAddr:"123"`, but I don't know enough about the topic to understand if that's a red flag or not.

Not reporting:
`{"id":33,"type":"EndDevice","ieeeAddr":"0x00158d000242c929","nwkAddr":123,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":2995,"3":29,"4":5032,"5":125,"6":[0,1],"8":516,"10":15289,"100":0},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1604632345769}`

Reporting:
`{"id":36,"type":"EndDevice","ieeeAddr":"0x00158d000242d7c0","nwkAddr":44400,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3005,"3":30,"4":5032,"5":88,"6":[0,1],"8":516,"10":0,"100":0},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1604665483732}`

McGiverGim, 2020-11-06:
nwkAddr is the network address. Each device must have one different. For the rest all seems ok.

alexcam1901, 2020-11-06:
I should have been more clear - the "123" seemed odd, the rest of my devices have 4 or 5 digit addresses.

I did some quick googling, it appears valid values can be 0x0000–0xfff7.  So nothing off there..

alexcam1901, 2020-12-07:
This is related to the non-standard Zigbee implementation from Xiaomi and inability of some routers to handle it. 

Replacing the Iris 3210-L routers with Tradfri routers solved the problem. 

See here:
https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

closed 2020-12-07 by alexcam1901