Koenkk/zigbee2mqtt #4802: CC1352P-2 touchlink failure

benv666, 2020-10-28:
## What happened

Tried to use touchlink to reset a Hue lightbulb. 
After seeing the initial touchlink message and a bunch of devices going offline due to ping failures there were a few errors in the log that I did not expect, followed by a pair of UnhandledPromiseRejectionWarning nodejs errors.

Logs here: https://hastebin.com/wayoyumedo.typescript

After the first try I disconnected and reconnected the adapter, restarted the zigbee2mqtt docker and tried again with the same result.
Logs here: https://hastebin.com/wuneletipu.lua

## What did you expect to happen

Touchlink to do its magic and reset my Hue bulb.

## How to reproduce it (minimal and precise)

Start touchlink reset, triggered using `mosquitto_pub -t "zigbee2mqtt/bridge/config/touchlink/factory_reset" -m ""`

## Debug info
Zigbee2MQTT version: koenkk/zigbee2mqtt:latest-dev, commit #82f3d7d
Adapter hardware: CC1352P-2
Adapter firmware version: CC1352P_2_20200805 (zStack3x0)

benv666, 2020-10-28:
Just tried the latest firmware version 20201026, same result (firmware seems to work fine otherwise in my short test).
Logs: https://hastebin.com/paveziyoxu.lua

benv666, 2020-10-28:
Created a log with herdsman debug enabled: https://hastebin.com/nohomukicu.sql

Interestingly when I wanted to create this log it didn't crash the adapter the first time: https://hastebin.com/semabacuzo.apache
(not that the touchlink worked on the bulb that's lying almost on top of the adapter)
I've only seen that happen once though, out of the 10-ish attempts now.

Koenkk, 2020-10-28:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/4729 (its a known problem in the Texas Instrument firmware).

closed 2020-10-28 by Koenkk