Koenkk/zigbee2mqtt #4803: State is not read when requested together with other properties.

TilmanK, 2020-10-28:
## What happened
Send a message to any get endpoint that contains "state" and any other property (i.e. "brightness"). Brightness will be requested and published, but state not. If only state is requested, it works.

Checked different devices and it seems to be device and vendor independent.

## Debug info
Zigbee2MQTT version: dev (https://github.com/Koenkk/zigbee2mqtt/commit/697955997aaa3b7853f68368a0887ae4f10f1cb8)
Adapter hardware: CC26X2R1
Adapter firmware version:

Koenkk, 2020-10-29:
Fixed, thanks for reporting

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-29 by Koenkk

TilmanK, 2020-10-29:
Thank you for the quick fix!