Koenkk/zigbee2mqtt #4804: SP-EUC01 not updating power

prutseltje, 2020-10-28:
## What happened
I've switched from a Texas Instruments CC2531 to a slaesh’s CC2652RB stick with latest firmare and had to re-add all my devices

## What did you expect to happen
All devices would work as before. I can switch the socket on and off, but the power value is stuck at 28.6

## How to reproduce it (minimal and precise)
Switch from CC2531 to CC2652RB stick and pair SP-EUC01

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_20200925.zip

Koenkk, 2020-10-28:
Try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

prutseltje, 2020-10-29:
Yes this fixes my issue!

closed 2020-10-29 by prutseltje

t-api-1, 2021-01-06:
I have the same issue but reconfiguring the device does not help, the same error persists. When I try to get power/energy/voltage readings from the device I get an error message of "No converter available for 'power' ()" or energy etc.

I am running the development version of z2m.

AdmiralStipe, 2021-01-15:
Similar case here, although power is reported correctly, but when I try to read voltage/current/..., I get "No converter available" message. Also in the Dashboard under Exposes only "State", "Power" and "Linkquality" are actually exposed, all other parameter are N/A.

mikesrus, 2021-01-23:
Same problem.
Reconfigure not help.

Milan6407, 2021-03-27:
Same problem.
Reconfigure not help.

> Similar case here, although power is reported correctly, but when I try to read voltage/current/..., I get "No converter available" message. Also in the Dashboard under Exposes only "State", "Power" and "Linkquality" are actually exposed, all other parameter are N/A.