Koenkk/zigbee2mqtt #4806: Make .yaml file inclusion in configuration.yaml generic

Koenkk, 2020-10-28:
## Problem
Currently it's possible to include other .yaml files for devices, groups and secret. However this does not allow to override any other properties. 

## Proposal
I like the proposal of @tricoos of https://github.com/Koenkk/zigbee2mqtt/issues/4697#issuecomment-716984907 a lot. Quoted:

@Koenkk Here is my proposal. I don't like if there is a single file that picks several attributes from various consistent blocks (like `mqtt` credentials but not the rest) and the `network_key` from the advanced section. However, splitting the security-related data is definitely useful.
Thus, what we actually do here - and I consider this very useful - is specifying a part of the values somwehere else. However, this might also confuse a bit so I suggest the following:

There is a new `includes` section that you can use to override any aspect of the configuration file as long as you use **the same hierarchy** as in the `configuration.yaml` file.

**Sample `configuration.yaml` (just as before, but with the new **includes** list)**

```
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
  ca: '/etc/ssl/mqtt-ca.crt'
  key: '/etc/ssl/mqtt-client.key'
  cert: '/etc/ssl/mqtt-client.crt'
  user: my_user
  password: my_password
...
includes:
  - secret.yaml
  - homeassistant.yaml
  - ..
```

**secret.yaml**
```
mqtt:
  user: my_real_user
  password: my_real_password
advanced:
   network_key: [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 14]
```

**homeassistant.yaml**
```
homeassistant: true
advanced:
  homeassistant_discovery_topic: 'homeassistant'
  homeassistant_status_topic: 'homeassistant/status'
  homeassistant_legacy_triggers: true
```
This allows you to even group related configuration values that reside in different parts of the config (like with `homeassistant`).

**Notes:**
- Regardless of the position of the includes part in the original config file it is replaced after parsing all of the data in the file
- With multiple include files specified they are replaced in the order the files are listed (i.e. if I override or define one value in the first includes file I can still replace it in the third one in the list). The last one always wins.
- This also means that the configuration file can now be split in as many parts as the user wants - a configuration value might be missing in **configuration.yaml**, but specified in one of the include files later.

Actually, as I have just split my Zigbee network into two I would now be able to use the same `configuration.yaml` but add a `custom.yaml` include file where I only list the differences (PAN id, network key, channel, ...)

sjorge, 2020-10-28:
We currently write group and device data like friendlyName to the config also...

Maybe we should revisit that also? As it will get tricky to pick the file to write too... like what if fileA sets some group stuff, fileB some devices stuff and fileC has other devices.... where do we write too?

Perphaps this data could just go into database.db, where IMHO it probably belongs... but how do we solve the 'easy' way to change a lot of stuff like friendlynames, filter_attributes, retain,... now I just edit the config and restart.

Although doing it over mqtt is not that much extra work.

Koenkk, 2020-10-29:
In this case we would write to the first file which has `devices` (first `configuration.yaml` next the order of `includes`)

closed 2020-12-06 by github-actions[bot]

Anonym-tsk, 2022-12-07:
Please reopen