Koenkk/zigbee2mqtt #4815: Deprecated click event - TRADFRI open/close remote

vannetta, 2020-10-29:
## What happened
I've a TRADFRI open/close remote; this is the payload:
`zigbee2mqtt:info  2020-10-29 12:48:41: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe083c73', payload '{"linkquality":26,"update_available":true,"click":"close"}'`
I've changed the automation using action instead of click; but the payload only shows the click.
I've tried to insert in the configuration.yaml 
```
devices:
  '0xccccccfffe083c73':
    friendly_name: 0xccccccfffe083c73_click
    legacy: false 
```
but when I check the configuration this is the message:
Component error: devices - Integration 'devices' not found.
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: zigbee2mqtt-edge
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II: CC2531
Adapter firmware version:

Koenkk, 2020-10-29:
To what `configuration.yaml` did you add this? I suspect you added this to the home assistant configuration.yaml but you should add it to the Zigbee2MQTT configuration.yaml instead.

netweaver1970, 2020-10-30:
I'm always on latest dev branch, updated multiple times a week. But I still had the issue. I
Will update the FW too and re-pair the outdoor sensor.

closed 2020-10-31 by Koenkk