Koenkk/zigbee2mqtt #4817: Slow network response after adding ikea trådfri lamps

taugusti, 2020-10-30:
## What happened
The problem is slow response. LED lights are really slow (or fails)
Everything worked fine while i only had sensors on zigbee2mqtt and the LEDs on the Trådfri gateway. 
Problems started when I moved the LEDs to Zigbee2mqtt.
Sensors (Ikea and Xiaomi buttons) works perfect.

** The setup **
35 zigbee devices connected:
 8x Trådfri on-off button
 18x trådfri led bulps
 2x trådfri motion sensors
 1x trådfri wireless dimmer
 1x trådfri remote control
 1x trådfri driver for led strips (10W)
 3x Xiaomi Aquara switches
 1x Xiaomi Aquara door/window sensor

**Eliminating home assistant:**
When sending a command diretcly from mqtt.fx ex:
zigbee2mqtt/0x000b57fffe121da3/set with payload {"state": "ON"}
The first command after a long silent period works fine. Maybe also the next - but after a few commands there is a delay (5-30 seconds) from command to response. Some lamps show this behavior from the very first try.
When home assistant responds to a button by turning on 2-4 lamps (both in and out via mqtt) the response is immediately very long

**Further observations**
if I try to "load map" in the home assistant zigbee2mqtt frontend, I get a lot of these errors:
zigbee2mqtt/bridge/logging: {"level":"error","message":"Failed to execute LQI for '0x000b57fffe1291e6'"}
It eventually loads the map - if it is correct or something is missing... I have no idea :-)

** What I have tried **
1. eliminate home assistant from the equation by controlling the LEDs from MQTT.fx - same result
2. replace CC2531 with conbee II (since I have >20 devices I expected that was the problem) - resulted in timeout errors:
  mqtt: zigbee2mqtt/bridge/log
  payload: {"message":"Publish 'set' 'state' to '0x000b57fffe121da3' failed: 'Error: Command 0x000b57fffe121da3/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"0x000b57fffe121da3"},"type":"zigbee_publish_error"}

## What did you expect to happen
no delays 👍 

## How to reproduce it (minimal and precise)
Install trådfri led lamps and the magic begins :-)

## Debug info
Zigbee2MQTT version: 1.5.0
Adapter hardware: CC2531 (tested with Conbee II also - same issue)
Adapter firmware version: 20190608
running on: Host: RPI4 with Home assistant 0.116.4 and the zigbee2mqtt plugin

** herdsman log **
https://pastebin.com/DhdHsAts

** Zigbee2mqtt home assistant config ***
```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.x.y.z'
  user: abc
  password: abcd
serial:
  port: /dev/ttyACM0
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
```

Koenkk, 2020-10-30:
Can you provide the herdsman debug logging with the Conbee 2 adapter? The CC2531 might not be powerful enough for this, the Conbee 2 should.

taugusti, 2020-10-30:
> Can you provide the herdsman debug logging with the Conbee 2 adapter? The CC2531 might not be powerful enough for this, the Conbee 2 should.

Sure - I will get it monday.

taugusti, 2020-11-02:
> Can you provide the herdsman debug logging with the Conbee 2 adapter? The CC2531 might not be powerful enough for this, the Conbee 2 should.

The log is available here: https://pastebin.com/8Sb1XA0x
The last enties in the log is where I click a button and hass sends a lights on command.
with the conbee it seems to work even worse...

@Koenkk : I hope you can help - it is much appreciated!

Koenkk, 2020-11-02:
@taugusti is your adapter connected through a USB extension cable?

taugusti, 2020-11-02:
> @taugusti is your adapter connected through a USB extension cable? (and thus positioned 20cm + from the pi 4)?

Yes - is is ~30 cm away - but the extension cord is 2m long and coiled up.

taugusti, 2020-11-02:
> > @taugusti is your adapter connected through a USB extension cable? (and thus positioned 20cm + from the pi 4)?
> 
> Yes - is is ~30 cm away - but the extension cord is 2m long and coiled up.

but - is that in anyway related to the problem? I receive actions from the buttons with no problems... also the ones furthest away.

taugusti, 2020-11-02:
Could my issue be related to this one: https://github.com/Koenkk/zigbee2mqtt/issues/4849
He describes similar behavior...

Today2000, 2020-11-02:
Hi,

Same problem since 1 month with last version of Zigbee2mqtt:

Zigbee network work very good, but if I power down one or more trådfri led bulps -with electric button-, few commands on zigbee network there is a delay (5-30 seconds) from command to respond and same log. Sometime zigbee network is complety down.

mqtt: zigbee2mqtt/bridge/log
payload: {"message":"Publish 'set' 'state' to '0x000xxxxxxxxxxx' failed: 'Error: Command 0x000xxxxxxxxxxxxx/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'","meta":{"friendly_name":"0x000xxxxxxxxxxxxx"},"type":"zigbee_publish_error"}

After if I  power up, after few second, all is working again.

I have flash my CC2531 with source routing exactly same problem

I have workaround for moment with:
add configuration.yaml:

advanced:
  availability_timeout: 60

And I test if topic 0x000xxxxxxxxxxx/availability == "online" before sent Set:ON or OFF command
Since 2 days with this workaround no problem in normal mode, no problem if one bulps is not power up on network

But somethings not working correctly

Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 20190619

taugusti, 2020-11-03:
@Koenkk - I just realized that after I went back to the conbee II adaptor, all the pulps show N/A in LQ. I assume that means that they are not available on the network?
[image]

Koenkk, 2020-11-03:
@taugusti no that's not what it means, are you able to control the devices? Note repairing is recommended when switching to a different adapter.

taugusti, 2020-11-03:
> @taugusti no that's not what it means, are you able to control the devices? Note repairing is recommended when switching to a different adapter.

no - I get timeouts when I try to control the lamps.
When I tried it last time (before posting) - I was able to control them some times (like with the cc coordinator).

It is just remarkable that inputs (ie buttons, motion sensors etc.) works great. Only the leds are not working. Does the log not give any hints? are there any other debugging I can do to get insight into the problem?

I will do the repair when I get the chance (it is in my summerhouse) and get back to you.

Koenkk, 2020-11-03:
@taugusti to dive a bit deeper we would need to sniff the traffic so we can see what happens: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

taugusti, 2020-11-03:
> @taugusti to dive a bit deeper we would need to sniff the traffic so we can see what happens: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ok - I will re-pair the devices and then sniff the traffic if the situation persists.
Thanks for your help @Koenkk !

taugusti, 2020-11-04:
hi @Koenkk . I have now deleted all devices and repaired a few of them:
[image]

Already with this few lamps in the system, problems began. Some lights work fine - some don't. all lamps are in range, paired just before they began to behave strange.

Ex. this lamp "indkørsel hoveddør 1" doers not work and gives timeouts:
```
Zigbee2MQTT:debug 2020-11-04 19:21:36: Received MQTT message on 'zigbee2mqtt/indkørsel hoveddør 1/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2020-11-04 19:21:36: Publishing 'set' 'state' to 'indkørsel hoveddør 1'
Zigbee2MQTT:error 2020-11-04 19:22:37: Publish 'set' 'state' to 'indkørsel hoveddør 1' failed: 'Error: Command 0x000b57fffeaf39b2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:debug 2020-11-04 19:22:37: Error: Command 0x000b57fffeaf39b2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
    at DeconzAdapter.<anonymous> (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:552:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)
Zigbee2MQTT:info  2020-11-04 19:22:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'indkørsel hoveddør 1' failed: 'Error: Command 0x000b57fffeaf39b2/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"indkørsel hoveddør 1"},"type":"zigbee_publish_error"}'
```

I will go for zigbee sniffing tomorrow.

taugusti, 2020-11-07:
> > @taugusti to dive a bit deeper we would need to sniff the traffic so we can see what happens: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
> 
> ok - I will re-pair the devices and then sniff the traffic if the situation persists.
> Thanks for your help @Koenkk !

Hi @Koenkk 
I have problems with the sniffing. When I run the zboss snifffer it says "can't start pcap logger":
[image]

I installed wireshark with all the options selected for winpcap, npcap, usbpcap. I have tried with both 32 and 64 bit versions of wireshark... there are no results to the error message on google...
I have not yet installed older versions of wireshark which will be next step - but it would be great to know if I am making an obvious mistake or if there is a specific version of wireshark I should use...?

Koenkk, 2020-11-07:
@taugusti sorry but I cannot help here, I have no experience with the Zboss sniffer software (I sniff on MacOS with just wireshark).

taugusti, 2020-11-10:
Ok, I realized I did not need wireshark to do the logging - and that the pcap files can be opened in wireshark later...
So here we go.
button_1.pcap: a Xiaomi button that is connected to a non-zigbee lamp. Works fine.
button_2.pcap: a Trådfri button that is connected to a zigbee lamp. does not work. 
light_1.pcap. trying to turn various zigbee lamps on an off from the zigbee2mqtt ui in home assistant. some works and some does not.
pairing_1.pcap: pairing a new lamp. takes very long time but succeeds at last.
[Downloads.zip](https://github.com/Koenkk/zigbee2mqtt/files/5515405/Downloads.zip)
I hope you can see the issues in the log @Koenkk 
You help is much appreciated.

Koenkk, 2020-11-10:
1. With what coordinator/firmware is this? 
2. button_2: if I understand correctly, you receive no event at all at the Zigbee2MQTT side?
3. light_1: I see the coordinator does not have routes and therefore does many route requests which puts heavy stress on the network. Need answer to question 1 to continue on this.
4. I don't see a Zigbee pairing sequence in this file. Could be that the sniffer was not in range of the paired device/router where it was paired to. A typical Zigbee pairing sequence starts with a Beacon -> Beacon response -> Associaton request -> Association response -> Transport key.

taugusti, 2020-11-11:
Hi @Koenkk 
ad 1: This is with the conbee II coordinator; firmware is deCONZ_ConBeeII_0x26660700.bin.gcf (7/oct/2020)
ad 2: no sorry - that was a clumsy description. I see that the sensor (trådfri button) is received in home assistant - but I see no response in the lamp.
ad 3: I this this is the core of the problem. my unscientific feeling was that the router effect is not working / configured correctly. 
ad 4: I will try this again tomorrow - it was very unreliable / slow to onboard devices when I was sniiffing so perhaps I was too impatient.
I onboarded two devices that are still shown as unknwon devices as can be seen here. Also note that the battery level from the trådfri button is shown as "?". really strange.
[image]

taugusti, 2020-11-12:
Ok - tried the onboarding again today
a lot of succesfull pairings (and a few failed) - but a clear picture that the distance to the coordinator is important. as if the relay in the network does not work as it should.
generally, it worked better today after I removed and re-paired the devices that did not work so well...
I hope the log files make the picture clearer.
please note - I updated to 1.6.1 in hass before the test.
[Downloads.zip](https://github.com/Koenkk/zigbee2mqtt/files/5532603/Downloads.zip)

Koenkk, 2020-11-12:
@chrishae do you know why the Conbee II requests so many routes? (does it have source routing enabled?). I feel somehow your problem is related to interference. Make sure to use a silent Zigbee channel: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html#reduce-wifi-interference-by-changing-the-zigbee-channel

ChrisHae, 2020-11-13:
Yes the FW version0x26660700 has source routing enabled. Is this related to the problem? It seems many users have those problems recently with conbee or cc_Stick.

taugusti, 2020-11-13:
> @ChrisHae do you know why the Conbee II requests so many routes? (does it have source routing enabled?). I feel somehow your problem is related to interference. Make sure to use a silent Zigbee [...]

Ok - I will try to change channel on monday.

taugusti, 2020-11-14:
> Yes the FW version0x26660700 has source routing enabled. Is this related to the problem? It seems many users have those problems recently with conbee or cc_Stick.

Is there a functional / beta firmware with source routing disabled I can use to test / get loggings of the impact it has?

ChrisHae, 2020-11-17:
The  latest FW without sourcerouting is
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26580700.bin.GCF
as far as I know. It is from may this year and there where many other fixes in later versions.

taugusti, 2020-11-17:
> Ok - I will try to change channel on monday.

ok @Koenkk 
I deleted all devices and then changed to channel 25. 
I did not succeed to pair a single device on that channel :-(
So I changed to channel 15. Managed to pair one single bulp - and after pairing I could not control it. Please see the attached log 
[pair on ch25 and 15.zip](https://github.com/Koenkk/zigbee2mqtt/files/5554258/pair.on.ch25.and.15.zip)
now going back to 11 and then downgrading conbee firmware.

taugusti, 2020-11-17:
> The latest FW without sourcerouting is
> http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26580700.bin.GCF
> as far as I know. It is from may this year and there where many other fixes in later versions.

Hi @ChrisHae 
Well - that did not go so well :-(
now I am back on channel 11 and old fw (your link) on the conbee.
And unable to pair any devices.
nothing shows up in mqttfx and nothing in the hass plugin.
but the pcap-file is not empty - so something have happened...
the pcap file is attached here.
I will now go back to new conbee on ch 11 to see if I can make anything work again.
[pair on old conbee fw ch11.zip](https://github.com/Koenkk/zigbee2mqtt/files/5554388/pair.on.old.conbee.fw.ch11.zip)

taugusti, 2020-11-17:
After further attempts at getting it to work, I realized that the last event in the log was always the join:true event. Nothing after that.
I eventually reinstalled the plugin, configured it from scratch again. I was able to join a couple of buttons but the first lamp had problems during interview. Unfortunately I did not get the zigbee log (only the landsman log - attached) because I had to leave.
I will go back tomorrow and give it a last chance. I need it to work on friday so I will have to revert to the old solution with the ikea gateway if it does not work... :-(

I am really sad if I have to abandon zigbee2mqtt as I think it is a brilliant solution - and when it worked, it worked great!
/th

taugusti, 2020-11-18:
Finally some progress
Not that I know exactly why it works better now - but it does.
I removed the plugin, rebooted, reinstalled
installed a conbee II with the old fw (deCONZ_ConBeeII_0x26580700.bin.GCF)
configured for channel 15.

With a couple of reboots during the process, and remove / re-pairs of some devices, I managed to add 39 devices.
It is not lightning fast - but not nearly as slow as with the cc2531 I used when I started the thread. And usable.
During the process I still saw a few timeouts, but it seems to be fairly stable now.

I hope it will remain stable.

Apparently the fw without source routing works better...
I have logged traffic on the network after being completely populated and configured.
I hope you @Koenkk can extract some useful information about the differences between before and now.
[operation of light and buttons after 39 device pair - old conbee fw.zip](https://github.com/Koenkk/zigbee2mqtt/files/5563051/operation.of.light.and.buttons.after.39.device.pair.-.old.conbee.fw.zip)

the only problems I saw during the test was that the "sofa 1" and "sofa 2" devices were really slow the first time.

Thanks for your help. I hope that you can use the information attached and in any case, I suggest that we leave the case open for a few days before closing it.

Best regards /th

taugusti, 2020-11-22:
After testing over the past days, I can say that using conbee II is much better than the cc2531. Changing to channel 15 and the older conbee firmware made a clear difference. 

I still see slow response. But the delays are shorter. 
I also see lack of response (ex. only one of 3 lamps turning on). But again not ast often as before.

I have 41 devices online, most of them unable to see the coordinator directly (relay required). I do not know if that is a part of the problem.

Unfortunately, it appears that the the logging and zigbee sniffing have not given any que to the bug.

I will close the thread now and hope that a future update will fix my problems.

closed 2020-11-22 by taugusti