Koenkk/zigbee2mqtt #4818: LAUNCHXL-CC26X2R1 works great- but at times needs powercycle

shahglobal, 2020-10-30:
Hi,
I have LAUNCHXL-CC26X2R1 adaptor installed in NUC pc for homeassistant. It works great and flawless most of the time but it does stop responding at times while everything works but Zigbee and a pc reboot do not help unless I turn the whole thing off and back on. Please advise do I need an external hardware watchdog for the power cycle or it can be fixed with a software update.
Thanks for this great integration always appreciated.
Shah

Koenkk, 2020-10-30:
Please provide your `configuration.yaml` and the herdsman debug logging the moment it stops working.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

shahglobal, 2020-10-30:
Thanks for the feedback sir, I will load your newer firmware later today as well and will follow the debug logging, hopefully, it will not put too much load on the normal operation or better would see if with newer firmware the behavior goes away if not then will attempt the debug logs collection and will post here.

pedrolamas, 2020-11-15:
I've been having a similar experience lately with my LAUNCHXL-CC26X2R1... my z2m just stops working after a couple of hours/days (the docker container never stops, but there's no messages flowing), but shutting down and power-cycling the computer (in my case, a NanoPi M4V2) fixes it.

I did spot this on the logs:

```
(node:16) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - extNwkInfo after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I'm using the latest stable firmware 20201026, but I will give a try to the new 20201113 and see if that helps!

shahglobal, 2020-11-15:
I spotted this behaviour again in even using fw 20201026, unfortunately, logs were not enabled that moment, now that I have enabled logging, will see if I can capture it if this happens in coming days. Kindly keep this open for now. By the way when the coordinator fails, its only Zigbee network that goes down, every other IoT devices RF and WIFI and hardwired ones keep working. And the fix is a power cycle, not a reboot. Standby for logs.
Kind Regards

pedrolamas, 2020-11-15:
I just had another failure, so I took the opportunity to flash 20201113 and enable full debug logging + herdsman logging and will report back if/when it fails again.

pedrolamas, 2020-11-16:
20h now since flashing 20201113, network going very well and no major errors in logs (filtering for "error" all I see is stuff like `zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x000d6f000be640b4:33842,233,0)`)

pedrolamas, 2020-11-21:
I'd seriously recommend anyone experiencing this issue with this adapter to try the new 20201113 firmware, as I can attest that this made a huge difference in performance and stability of my zigbee network!

I previously had to occasionally reboot (full power down) and sometimes even reflash the adapter mostly because of BUFFER_FULL errors, but I've had zero problems since I started using this beta firmware!

shahglobal, 2020-11-21:
Just in time, it stopped working now, but thanks for the pointer I will upgrade the firmware first thing in the morning tomorrow.
Here is a portion of the log at the start of failed time not sure if it is of any help but moving on I will upgrade the firmware shortly.
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/5577402/logs.txt)
Kind Regards

Koenkk, 2020-11-21:
@pedrolamas that's very good to hear! thanks for your feedback :smile:

netweaver1970, 2020-11-21:
I today jumped to the Nov 13 beta FW as well, using it together with a (previously) problematic Hue Outdoor sensor and so far all fine. Of course it's still early days to 100% confirm it but so far all great. Thanks.

netweaver1970, 2020-11-23:
Mmm last night the Hue outdoor sensor left the network again. So no solution for that problem yet, not with latest beta FW and latest z2m. Sad ...

shahglobal, 2020-11-26:
Update:- 
This problem happened again last night even with firmware 20201113 and reboot did not fix the problem but the power cycle did. The whole ZigBee network ( maybe 50 devices) was down. The logs did not catch it there was only a one-day log folder only for today's date and only restarting info was there when I was restarting then back to normal logging of devices.
So please keep digging so that is could be resolved, please advise. or if some sort of watchdog runs on usb to power cycle the device? meanwhile when we look for core issue. Hopefully, someone smarter can spot this issue better fingers crossed.
Kind Regards

shahglobal, 2020-12-10:
Good day,
Here is the logs of today's Zigbee failure. The power cycle fixed it.
[error.txt](https://github.com/Koenkk/zigbee2mqtt/files/5670051/error.txt)

Thanks

Koenkk, 2020-12-10:
@shahglobal try to put the following in your configuration.yaml:

```yaml
advanced:
  adapter_concurrent: 1
```

shahglobal, 2020-12-19:
Good day
I reinstalled the whole system and not all the ZigBee devices were even connected yet it worked for some time and then it failed this morning, I have attached the logs for your review since this was a new installation I missed the adapter_concurrent flag.
you can see that the failure started at 03:54:27 in logs.
Kind Regards,
[errordec19.txt](https://github.com/Koenkk/zigbee2mqtt/files/5719925/errordec19.txt)

Koenkk, 2020-12-20:
@shahglobal could you provide the herdsman debug log from starting z2m till the error?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

shahglobal, 2020-12-30:
Good morning,
Here is a complete log file before my watchdog rebooted the server after Zigbee
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5754763/log.txt)
 failure, kindly review it. This happened this am.
Cheers

Koenkk, 2020-12-31:
@shahglobal unfortunately this is not the herdsman debug logging. Note that this is only logged to STDOUT, not the log files.

shahglobal, 2020-12-31:
Thanks for the feedback sir, looks like the logs that are required are only available at the log tab of the application, and as I understand it is only one page of data that gets refreshed and also gets lost at reboot. Is there any way to redirect that log to a file so that I can capture it and send it here?
Kind Regards

Koenkk, 2021-01-01:
@shahglobal if you are running the HA addon, you can ssh into the machine and do a `docker logs -f` on the Z2M container.

shahglobal, 2021-01-01:
Hi,
When I run "docker logs -f d866eb094586 > z2m.log" it logs the same info as regular logs I sent before, the command console does show up herdsman debug but the file it creates receives normal log data, not herdsman. Not sure why this command is not dumping the info to the file while displaying it correctly.

Koenkk, 2021-01-01:
@shahglobal I think you need to run the edge version of the addon

closed 2021-02-09 by github-actions[bot]