Koenkk/zigbee2mqtt #4819: Problem with Legrand Netatmo wired shutter switch 067776

Riseryn, 2020-10-30:
# Bug Report
## What happened
The parameter "position" only work with 0(closed) and 100(open) value.
Other value doesn't work.
## What did you expect to happen
I expect position: 50 to half open the store
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.15
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-10-31:
@cemonneau  does this work for you?

Riseryn, 2020-11-04:
using mqtt.fx 1.7.1 to retrieve informations.
when I manually activate the shutter control button I get 3 messages which are:

zigbee2mqtt/shutter
{
  "action" : "moving",
  "linkquality" : 26,
  "position" : 50
}

{
  "linkquality" : 26,
  "position" : 50
}

{
  "linkquality" : 28,
  "position" : 50
}

 "position" : 50 corresponds to the last requested position and not to the actual position of the shutter.

Riseryn, 2020-11-09:
No ideas about this issue?

eRJe79, 2021-04-28:
@Riseryn did you get any further with this?

Riseryn, 2021-04-29:
Hi,
absolutely not.
Im unable to move at a specified position. Only open and close.

eRJe79, 2021-04-30:
@Koenkk can we reopen this issue?

closed 2021-05-08 by github-actions[bot]

alapasset, 2022-05-02:
@Koenkk  can we reopen this issue?

cyrilphoenix71, 2022-05-30:
Hello I have the same problem with white version 067726, use position slide bar doesn't work, I have always 50 in position state.
Could you help us ? @Koenkk

Koenkk, 2022-05-31:
For this device standard zigbee commands are used. First impresion is that this device simply doesn't support intermediate positions (but someone should check this with the original hub)

benjR, 2022-06-14:
Same issue for me, unfortunately I don't have the Legrand Hub to test.

cyrilphoenix71, 2022-06-14:
I hear about new firmware on this device that open functionnality but I don't know how can I do that

benjR, 2022-06-14:
@cyrilphoenix71 do you have a link?

bemble, 2022-06-18:
> I hear about new firmware on this device that open functionnality but I don't know how can I do that

Where did you get this? The device is not listed on Legrand compatible list..: https://developer.legrand.com/documentation/compatible-product/

bemble, 2022-06-19:
As said on #12655, not even possible on legrand app, we have to wait for firmware upgrade that brings it.