Koenkk/zigbee2mqtt #4821: Home Assistant Scenes Discovery

kiall, 2020-10-30:
## Describe the solution you'd like

The new Scenes support is great, but configuring Home Assistant to use them reminds me of the bad old days :D We should add discovery support.

I've submitted a PR to HA to add MQTT Scenes: https://github.com/home-assistant/core/pull/42639 - I'd love some review of that PR to ensure it'll suit current and future plans. For now (and at least IMO), Home Assistant only needs to be able to activate/recall a scene - so that's what I've implemented.

## Describe alternatives you've considered

Manually configuring scripts and buttons and so on in HA to configure scenes.

## Additional Context

Related to https://github.com/Koenkk/zigbee-herdsman-converters/pull/1531

Koenkk, 2020-10-31:
Looks good! Once it's implemented we can look into the Z2M implementation.

kiall, 2020-12-02:
Still WIP over at HA: https://github.com/home-assistant/core/pull/42639

kiall, 2020-12-03:
And, support has merged to HA & is included in the just announced v1.0.0 rc.

sjorge, 2020-12-03:
Just wonder if there is also view and remove support on the hass side, we don't have it yet in z2m... but should probably at least add scene_remove and maybe scene_remove_all, it's on my way to long list of things to do -_-

kiall, 2020-12-03:
On the HASS side, no - just the ability to activate a scene for now. I don't think it supports anything other than activating an external scene, but I could be wrong on that!

sjorge, 2020-12-03:
Ah so only scene_recall! I guess pushing complex scenes to the group/device might be a bit complicated to implement yeah.

kiall, 2020-12-03:
Yep - I figured that would be by far the most useful thing, so it's all I set out to do :) Though, on the way through doing it, I've figured it's likely not possible to add other actions anyway!

MattWestb, 2020-12-03:
Good to implanting in HA is save scene / groupe command being sent to Z2M. Also for the default scene groupe 0x0000 (and IKEA remote 0xff09).

ndfred, 2020-12-23:
For those wanting to try scene integration in HA, it is not in the official release just yet. Adding this section causes an error message `Setup failed for scenes: Integration not found` in the HA logs:

    scenes:
      - platform: "mqtt"
        name: Living Room Bright
        unique_id: living_room_bright
        command_topic: "zigbee2mqtt/living_room/set"
        payload_on: '{"scene_recall": 4}'

      - platform: "mqtt"
        name: Living Room Dimmed
        unique_id: living_room_dimmed
        command_topic: "zigbee2mqtt/living_room/set"
        payload_on: '{"scene_recall": 5}'

      - platform: "mqtt"
        name: Living Room Nightlight
        unique_id: living_room_nightlight
        command_topic: "zigbee2mqtt/living_room/set"
        payload_on: '{"scene_recall": 6}'

I'll wait until the next HA release comes out and try this again, apparently the next one should be on the [first Wednesday in January](https://www.home-assistant.io/blog/2020/12/13/release-202012/).

kiall, 2020-12-23:
MQTT scenes support has definitely landed in the 2020.12 release, no need to wait till January!

Your config example looks correct, I don't know why you would be seeing that error. Are you sure HA's MQTT is setup correctly?

ndfred, 2020-12-23:
I am on 2020.12.1 and am getting this error message though, and indeed your changes are [listed in the changelog](https://www.home-assistant.io/blog/2020/12/13/release-202012/). Found why that was the case, I had this default section in my config file that I never really used since this is my first time leveraging HA scenes:

    scene: !include scenes.yaml

This means I would have redefined the scene section which YAML wouldn't have been keen on, but really my mistake was to introduce a `scenes` rather than `scene` section. I just moved my stuff over to `scenes.yaml` and there is :

    - platform: "mqtt"
      name: Living Room Bright
      unique_id: living_room_bright
      command_topic: "zigbee2mqtt/living_room/set"
      payload_on: '{"scene_recall": 4}'

    - platform: "mqtt"
      name: Living Room Dimmed
      unique_id: living_room_dimmed
      command_topic: "zigbee2mqtt/living_room/set"
      payload_on: '{"scene_recall": 5}'

    - platform: "mqtt"
      name: Living Room Nightlight
      unique_id: living_room_nightlight
      command_topic: "zigbee2mqtt/living_room/set"
      payload_on: '{"scene_recall": 6}'

If I go to Configuration > Scenes in HA they're all there and work really well! Thanks for the pointers, and sorry for my silly mistakes :)

RobertMe, 2021-09-07:
I've been searching for this feature and could only find this ticket. As it's closed without any PR or commits being linked I think this feature request has gone stale and isn't implemented, am I correct?

The part on Home Assistants end obviously has been added, but nu discovery support has been added to Zigbee2mqtt? As this would be my last hurdle to ditch the Hue Bridge.

closed 2021-09-15 by github-actions[bot]

jkaberg, 2021-11-05:
Dumb bot, no cookie for you

GhostPunk, 2022-01-06:
Is this feature still on the roadmap? Manually defining scenes work like a charm since HASS merged the PR but obviously would be better if it would be automatic.

RobertMe, 2022-01-06:
I might be able to work on this.

But checking the code I'm really confused on how to get a proper list of scenes and what they are bound to. I did find  `Endpoint.meta?.scene` and `Group.meta?.scene` which could be used, but this means having to iterate over all devices and groups, getting their scenes, splitting the "code" (being `<scene id>_<group id>`) checking if the scene is for a group (`group_id` not being `0`) and using the groups friendly name in that case or if the scene is for the device itself (`group_id` being `0`) using the friendly name of the device itself which would be the topic to invoke. And what further complicates this is that when changing the group or device name the topic will obviously change as well so in that case the HA discovery must be re-done too. Furthermore the scene name is stored as part of the scene metadata on the device/endpoint. So it even is possible that a group with two devices will have a different scene name set, so which name should be used then? As IMO this should be the auto generated entity name in HA.

timdonovanuk, 2022-12-16:
Sooo...is this added or going to be added? Really confusing info in this thread! It doesn't seem like z2m scenes are mqtt auto discovered by HA yet.

chinconl, 2023-01-05:
Could you copy the behavior from Hue/HA? It works fine there.

ukmgranger, 2023-02-24:
I assume that this never got added / implemented?  Would be really useful to be able to see / use scenes easily in HA.

radclif, 2023-10-03:
I would also love it if Z2M scenes where synced to HA and not having to redefine every scene in the YAML. I have 200+ zigbee devices and it's a big pain to manage it all. I am a PHP programmer (I know some Python) and I would love to help on this but I don't know where to start.

PR #19838 (2023-11-26): Home Assistant discovery improvements (including scenes)
- Adds device information for groups.
- Adds Home Assistant auto-discovery for group scenes. Fixes #4821, #13937, #18481.
- Adds the optional `object_id_postfix` key in the discovery payload to customize the `entity_id` in Home Assistant.