Koenkk/zigbee2mqtt #4824: Nue 3a single gang switch not responding to HA commands

grantleyring, 2020-10-31:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-10-31:
Can you provide the Zigbee2MQTT debug log when controlling the device via HA?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

grantleyring, 2020-11-01:
Thankyou for being so quick absolutely fantastic. I think I am supposed to supply a link to this file, but I don,t know how to do that so I added a copy of the log in a file here, I have editied the file at the bottom to show where I controlled the switch from HA. It also shows the other single gang switch that has the same issue.

Hope this is what you want and thanks again.

[Zigbee2mqtt_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5470009/Zigbee2mqtt_log.txt)

Koenkk, 2020-11-01:
```
Zigbee2MQTT:debug 2020-11-01 11:11:25: Received MQTT message on 'zigbee2mqtt/0x00158d000476b312/set' with data 'True'
Zigbee2MQTT:error 2020-11-01 11:11:25: Invalid JSON 'True', skipping...
Zigbee2MQTT:debug 2020-11-01 11:11:26: Received MQTT message on 'zigbee2mqtt/0x00158d000476b312/set' with data 'False'
Zigbee2MQTT:error 2020-11-01 11:11:26: Invalid JSON 'False', skipping...
```

HA does send an incorrect payload when controlling the device, could you make a small gif/video how you control the device via HA?

grantleyring, 2020-11-01:
Thanks @Koenkk 

Hope these files are OK.
I have simply created an entities card with a single switch pointing to the entity. Notice the switch does not remain on and goes of by itself after approx 2 seconds.

[image]
[image]
[image]

Koenkk, 2020-11-01:
Can you post the debug log of the Zigbee2MQTT startup? (+- first 30 seconds?)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

grantleyring, 2020-11-01:
This is the Zigbee2mqtt configuration file and the logfile following a restart of HA.
If it would make life any easier for you, let me know and I will create an account for you to login as I use nabu casa which I am assuming you can use to get to my system (as long as you dont laugh at my novice setup ).

Regards,
Grant

[Zigbee2mqtt_config.txt](https://github.com/Koenkk/zigbee2mqtt/files/5470844/Zigbee2mqtt_config.txt)
[Zigbee2mqtt_startup.txt](https://github.com/Koenkk/zigbee2mqtt/files/5470845/Zigbee2mqtt_startup.txt)

Koenkk, 2020-11-01:
@grantleyring I think you grabbed the log from the page as the log does not start at the Zigbee2MQTT start. I think you can grab the complete log file from the `data/log` directory (which is next to the Zigbee2MQTT configuration.yaml file).

grantleyring, 2020-11-01:
Hope this one is better.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5470897/log.txt)

Koenkk, 2020-11-01:
This looks like a Home Assistant bug to me, the discovery information states to use `ON`/`OFF` but `true`/`false` is sent. I have no clue why this does not occur people (for me this is working fine).

Discovery payload: `{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0x00158d000476b312/set","device":{"identifiers":["zigbee2mqtt_0x00158d000476b312"],"manufacturer":"Nue / 3A","model":"Smart one gang wall switch (HGZB-41)","name":"0x00158d000476b312","sw_version":"Zigbee2MQTT 1.15.0"},"json_attributes_topic":"zigbee2mqtt/0x00158d000476b312","name":"0x00158d000476b312_switch","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x00158d000476b312","unique_id":"0x00158d000476b312_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}`

@emontnemery  is this a known issue?

grantleyring, 2020-11-01:
This got above my pay grade after "Starting Zigbee2MQTT version 1.15.0"  (LOL) is there anything I can do?
Regards,
Grant.

emontnemery, 2020-11-01:
From the issue description it seems like @grantleyring has manually configured the switch instead of relying on MQTT discovery.

There are a few things about that which are not obvious:
- `device` configuration is not possible through `configuration.yaml`, this is mentioned in the HA docs
- YAML is **extremely** particular in how it treats unquoted strings. One gotcha is that the following unquoted strings will be parsed as booleans:  y|Y|yes|Yes|YES|n|N|no|No|NO|true|True|TRUE|false|False|FALSE|on|On|ON|off|Off|OFF

grantleyring, 2020-11-02:
Hi @emontnemery ,

You are absolutely correct I have manually configured these switches which I do in switches.yaml. I do this only because after attempting to enable auto discovery and multiple reinstallations of both MQTT and Zigbee2MQTT I was unable to get auto discovery working (in spite of reading the documation several times). I assume this is ignorance on my part but I do not want to consume the valuable and busy time of people like yourselves who have far more important problems to solve.

So my process is to read the MQTT (Listen to a topic) in the pairing phase convert the json to yaml via an online program and put this into my switches file. It appears to work for all but the single gang switches.

Is the solution as simple as placing quotes within the switch definition?

Regards,
Grant.

grantleyring, 2020-11-02:
Yes the solution is to Quote the payload parameters.  Tried it and all is good
My sincere thanks to all that helped in this great stuff guys,
Reagrds,
Grant.

closed 2020-12-10 by github-actions[bot]