Koenkk/zigbee2mqtt #4827: OSRAM SMART+ 4x switch version AB371860155|x1117

Plazmin, 2020-10-31:
This switch is recognized as "Switch 4x EU-LIGHTIFY" model: '4058075816459' Short and long presses work properly but the switch does not report battery status. Where is the problem? Do you need to configure something extra?
[image]

Koenkk, 2020-11-01:
Battery can take up to 24 hours to report after pairing, did you wait this long?

Plazmin, 2020-11-01:
Yes. I waited a few days, but there is no information about the battery status. Osram SMART + Motion Sensors (AC01353010G) send information about the battery every time. Is it different with the 4x switch? Maybe the switch only reports if the battery is low?

Koenkk, 2020-11-01:
Well found that is indeed a battery report, fixed this.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Plazmin, 2020-11-01:
Thank you for your quick reply and your work. I have a question. Will this patch also work with my version of the switch, i.e. AB371860155|x1117 ?

Koenkk, 2020-11-01:
Yes

Plazmin, 2020-11-04:
Now, after changing the firmware from the dev branch and pairing the switch again, everything works fine. Thank you Koenkk.

closed 2020-11-04 by Plazmin