Koenkk/zigbee2mqtt #4829: zigbee2mqtt autostart with systemd fails on first try

Invento3D, 2020-10-31:
I have set up the following service to start at boot:
[image]

However, it doesn't work. After rebooting my system, I get the following in the zigbee2mqtt log:
[image]

The thing is, if I now start zigbee2mqtt again manually via `sudo systemctl start zigbee2mqtt.service`, it starts up correctly.
As you can see I already tried to delay the autostart of the service a bit by `ExecAfter`, but that didn't help.

Koenkk, 2020-11-01:
Please provide the herdsman debug log of the startup error.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Invento3D, 2020-11-01:
I found the problem.
The service was not even auto-starting anymore.
After re-enabling it with `sudo systemctl disable zigbee2mqtt.service && sudo systemctl enable zigbee2mqtt.service` it works again.

closed 2020-11-01 by Invento3D