Koenkk/zigbee2mqtt #4836: After upgrading to version 1.16.0, the Web UI/Frontend does not open anymore

gadric, 2020-11-01:
## What happened
After upgrading to version 1.16.0, the Web UI/Frontend does not open anymore. 

## What did you expect to happen
The Frontend/Web UI should open as in the previous version (e.g., 1.15.0)

## How to reproduce it (minimal and precise)
1. Install version 1.16.0
2. Restart Host and Home Assistant
3. Open Web UI via Addon site or via sidebar
4. Page remains blank ..

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2531 with external Antenna
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30

tomaae, 2020-11-01:
Same, except I cannot control anything zigbee related. Everything shows as available in HA, but nothing reacts.

gadric, 2020-11-01:
Just tested, my MQTT sensors and also my Power Plug works over zigbee2mqtt. Only the Frontend does not work ...

skumka, 2020-11-01:
I can control mostly nothing zigbee related. zigbe2mqtt seems to work itself (except  map is not readable at all due to changes.)
All items in UI show up as UNAVALILABLE and cannot be controlled.
[image]

tomaae, 2020-11-01:
I have removed z2m, deleted all configuration and rebuild everything from the start. now everything works, including frontend.
looks like something messes up during upgrade

skumka, 2020-11-01:
@tomaae  tomaae did you rediscover all zigbee sensors?
or you copied all from backup?
could you please list what did you do exactly? step by step

tomaae, 2020-11-01:
I just copied devices.json so I dont have to rename them, but had to join in all devices one by one.

skumka, 2020-11-01:
oh NO... thanks @tomaae 

@Koenkk how to solve the issue without need to rediscover everyting?

tomaae, 2020-11-01:
yeah, took me almost 2 hours, not fun. but I had to fix it fast

gadric, 2020-11-01:
@tomaae oh no ... I hope this is not the only solution. I would not like to rejoin all my devices ... there must be other solutions ...

Koenkk, 2020-11-02:
- Are you all using the Home Assistant addon?
- Can you provide the herdsman debug logging of the startup?

skumka, 2020-11-02:
I do no use the HomeAssistant add-on. I have HA in docker. Zigbee and Mosquitto on bare metal.
I have solved the problem by removing the whole zigbee2mqtt folder. I made a fresh reinstall and copied the data folder from backup.
Definitely problem is somewhere in migration. I noticed that 2 od my Zigbee devices were working correctly after migration but all others were unavailable.

gadric, 2020-11-02:
I am using the Home Assistant Addon. 

This is what the log shows at startup: 

```

Zigbee2MQTT:info  2020-11-02 08:30:40: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-11-02 08:30:42: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-02 08:30:42: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"}'

```

But I guess you need other information right? How do I get to the debug of herdsman?

Koenkk, 2020-11-02:
@gadric what shows after this?

gadric, 2020-11-02:
Zigbee2MQTT:info  2020-11-02 11:55:16: xiaomi_mijia_window_left (0x00000000000000000000): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-02 11:55:16: xiaomi_mijia_balcony_door (0x00000000000000000000): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-02 11:55:16: xiaomi_mijia_window_right (0x00000000000000000000): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-02 11:55:16: ikea_tradfri_signal_repeater (0x00000000000000000000): E1746 - IKEA TRADFRI signal repeater (Router)
Zigbee2MQTT:info  2020-11-02 11:55:16: xiaomi_mijia_smartplug_rpi4 (0x00000000000000000000): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
Zigbee2MQTT:info  2020-11-02 11:55:16: xiaomi_aqara_door (0x00000000000000000000): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-02 11:55:16: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2020-11-02 11:55:16: Started frontend on port 8099
Zigbee2MQTT:info  2020-11-02 11:55:17: Connecting to MQTT server at mqtts://xxx.xxx.xxx.xxx:8883

Koenkk, 2020-11-02:
@gadric and what is logged when you control a device?

gadric, 2020-11-02:
This is what happened after I turned the switch to `OFF`

```
Zigbee2MQTT:info  2020-11-02 19:48:32: MQTT publish: topic 'zigbee2mqtt/xiaomi_mijia_smartplug_rpi4', payload '{"consumption":1.19,"current":0.0095,"energy":1.19,"linkquality":39,"power":2.18,"state":"OFF","temperature":27,"voltage":230}'
Zigbee2MQTT:info  2020-11-02 19:48:32: MQTT publish: topic 'zigbee2mqtt/xiaomi_mijia_smartplug_rpi4', payload '{"consumption":1.19,"current":0.0095,"energy":1.19,"linkquality":42,"power":2.18,"state":"OFF","temperature":27,"voltage":230}'
Zigbee2MQTT:info  2020-11-02 19:48:32: MQTT publish: topic 'zigbee2mqtt/xiaomi_mijia_smartplug_rpi4', payload '{"consumption":1.19,"current":0.0095,"energy":1.19,"linkquality":42,"power":0,"state":"OFF","temperature":27,"voltage":230}'
```

jlaitinen81, 2020-11-03:
Same problem here. All devices work properly, but the WEB UI is blank.

jlaitinen81, 2020-11-03:
It seems that this bug is somehow related to Ingress. I still have the problem with Safari browser, but with Chrome the UI works perfectly well.

gadric, 2020-11-03:
> It seems that this bug is somehow related to Ingress. I still have the problem with Safari browser, but with Chrome the UI works perfectly well.

Agree. Just verified. On my iOS device and in the iOS Home Assistant App, the Web UI is still blank. But in Chrome or Firefox on a computer it works.

diabl0, 2020-11-03:
> * Are you all using the Home Assistant addon?

HA and Z2M in docker containers.

`
services:
    homeassistant:
        container_name: home-assistant
        image: homeassistant/home-assistant:stable
        volumes:
            - ./config:/config
        environment:
            - TZ=Europe/Warsaw
        restart: always
        network_mode: host
        privileged: true

    zigbee2mqtt:
        container_name: zigbee2mqtt
        image: koenkk/zigbee2mqtt:1.15.0
        volumes:
            - ./zigbee2mqtt/data:/app/data
            - /run/udev:/run/udev:ro
        devices:
            - /dev/ttyACM0:/dev/ttyACM0
        restart: always
        network_mode: host
        privileged: true
        environment:
            - TZ=Europe/Warsaw
`

In my case Home Assistant had problem at least with lights and buttons (IKEA E1524/E1810 buttons and LED1903C5/LED1835C6 lights - solo and grouped). On buttons press automations last run in HA was updated, but lights didn't change. Controll via mqtt.publish service on topic: zigbee2mqtt/attic_lights/set works (not perfecty as i wasn't abble to set brightnes/temperature together with state: toggle, but it is topic for another issue that i will try to document when issues with 0.16 will be resolved). 

Didn't have time to check other devices as my wife started to complain about broken lights, and had to revert docker image to koenkk/zigbee2mqtt:1.15.0

DzianisFrydliand, 2020-11-05:
@Koenkk 
I have the same issue.
Please let me know which logs can help.

Here are some lines from Supervisor log:
`20-11-05 08:42:03 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
20-11-05 08:42:03 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
20-11-05 08:42:03 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
20-11-05 08:42:03 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
20-11-05 08:42:03 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]`

DzianisFrydliand, 2020-11-05:
Maybe this also could help:

`➜  ~ docker ps
CONTAINER ID        IMAGE                                                COMMAND                  CREATED             STATUS              PORTS                                                                NAMES
f4ebe465bd75        dwelch2101/zigbee2mqtt-armv7:1.16.0                  "/init ./run.sh"         3 hours ago         Up 3 hours          0.0.0.0:8485->8485/tcp                                               addon_7ad98f9c_zigbee2mqtt
84ff44686e93        homeassistant/raspberrypi3-homeassistant:0.117.4     "/init"                  3 hours ago         Up 3 hours                                                                               homeassistant
511c2aa5b761        homeassistant/armv7-hassio-supervisor:latest         "/init"                  26 hours ago        Up 26 hours                                                                              hassio_supervisor
d9397de48558        hassioaddons/sqlite-web-armv7:2.3.1                  "/init"                  2 days ago          Up 2 days           0.0.0.0:6220->6220/tcp                                               addon_a0d7b954_sqlite-web
dd1d74d66ff5        homeassistant/armv7-hassio-dns:2020.11.0             "/init"                  3 days ago          Up 3 days                                                                                hassio_dns
5006156b5cfb        hassioaddons/log-viewer-armv7:0.9.1                  "/init"                  3 days ago          Up 3 days                                                                                addon_a0d7b954_logviewer
2eb0d0d35454        homeassistant/armv7-addon-configurator:5.1.0         "/init"                  3 days ago          Up 3 days                                                                                addon_core_configurator
552395c86a3c        sabeechen/hassio-google-drive-backup-armv7:0.102.0   "python3 -m backup"      3 days ago          Up 3 days           0.0.0.0:1627->1627/tcp, 8099/tcp                                     addon_cebe7a76_hassio_google_drive_backup
8deee79c7dff        hassioaddons/ssh-armv7:7.7.0                         "/init"                  3 days ago          Up 3 days                                                                                addon_a0d7b954_ssh
06185f57d124        homeassistant/armv7-addon-mosquitto:5.1              "/run.sh"                3 days ago          Up 3 days           0.0.0.0:1883-1884->1883-1884/tcp, 0.0.0.0:8883-8884->8883-8884/tcp   addon_core_mosquitto
f9b8ba6ca589        homeassistant/armv7-hassio-multicast:3               "/init"                  3 days ago          Up 3 days                                                                                hassio_multicast
b18e3927ca80        homeassistant/armv7-hassio-cli:2020.10.1             "/init /bin/bash -c …"   3 days ago          Up 3 days                                                                                hassio_cli
ba626bfb4c36        homeassistant/armv7-hassio-audio:17                  "/init"                  3 days ago          Up 3 days                                                                                hassio_audio
1e645d23ad81        homeassistant/armv7-hassio-observer:2020.10.1        "/init"                  2 weeks ago         Up 3 days           0.0.0.0:4357->80/tcp                                                 hassio_observer`

Koenkk, 2020-11-05:
Possible the same issue as https://github.com/danielwelch/hassio-zigbee2mqtt/issues/460 , occurs when your MQTT server requires no authentication. Fixed in 1.16.1 and the dev branch (edge for hassio) in a few hours.

DzianisFrydliand, 2020-11-05:
Don't think it the same issue... I have MQTT login/pass set in my config:
`mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: XXX
  password: YYY`

DzianisFrydliand, 2020-11-05:
Update to 1.16.1 did not resolve the issue

fmarczin, 2020-11-06:
I have the same original issue as reported by @gadric, still with 1.16.1. Web UI works in Chrome, but not in Safari.

DzianisFrydliand, 2020-11-06:
> I have the same original issue as reported by @gadric, still with 1.16.1. Web UI works in Chrome, but not in Safari.

Same. Also UI doesn't work in Home Assistant app on iOS.

Koenkk, 2020-11-06:
I don't think this is directly related to Zigbee2MQTT but the ingress/Home Assistant addon, please open an issue here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

ciotlosm, 2020-11-09:
Apparently this (frontend blank in home assistant) is related to caching. I was able to reproduce this on my IOS setup in the native app, but clearing frontend cache from the application settings. 

I assume for those seeing the same problems in the browsers a clear cache for their hass url or even maybe a force refresh (shift + R / shift + F5) might do the trick.

gadric, 2020-11-09:
@ciotlosm: Thanks! For the iOS App I can confirm.

closed 2020-12-18 by github-actions[bot]