Koenkk/zigbee2mqtt #4837: Lubeez 12AB Zigbee dimmer not supported

worker72, 2020-11-01:
The zigbee AC dimmer [Lubeez 12AB](https://www.bol.com/nl/p/lubeez-zigbee-3-0-ac-dimmer/9200000130356125/?bltgh=pONgTAdW88K3IDap9MhK7w.tYKRVps-Tj87UDoXRKFPug_0_11.12.ProductPage) looks very similar to other AC dimmers (e.g. [ROBB](https://www.zigbee2mqtt.io/devices/ROB_200-004-0.html)), but it is not supported by zigbee2mqtt.
I tried to follow all documentation. I copied the herdsman devices.js file to my zigbee2mqtt folder, and added my device, copying the text of an existing device. 
The dimmer appears now in HA, but I only see linkquality and an on/off switch, I'm not able to set the brightness.
Here is what I put in devices.js:
```
	// LUBEEZ
	{
		zigbeeModel: ['LUBEEZ-12AB'], // 
		model: '12AB', // Vendor model number, look on the device for a model number
		vendor: 'Lubeez', // Vendor of the device (only used for documentation and startup logging)
		description: 'zigbee 3.0 AC dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
		extend: generic.light_onoff_brightness,
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await configureReporting.onOff(endpoint);
        },
    },
```
The log says "Successfully configured '0x680ae2fffe3a64af"

the relevant line in database.db:
`{"id":19,"type":"Router","ieeeAddr":"0x680ae2fffe3a64af","nwkAddr":62622,"manufId":4644,"manufName":"LUBEEZ","powerSource":"Mains (single phase)","modelId":"LUBEEZ-12AB","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LUBEEZ-12AB","manufacturerName":"LUBEEZ","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r47"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0008b87c25","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0008b87c25","endpointID":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r47","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1604243678535}`

What must I do to be able to set the brightness from HA?

## Debug info
Zigbee2MQTT version:  1.16.0
Adapter hardware: CC2531
Adapter firmware version:{'meta': {'maintrel': 3, 'majorrel': 2, 'minorrel': 6, 'product': 0, 'revision': 20190608, 'transportrev': 2}, 'type': 'zStack12'}

Koenkk, 2020-11-02:
Can you make a gif on what you see in Home Assistant? When the device is turned on it should show you a brightness slider.

worker72, 2020-11-02:
No, there is no slider

[image]

Koenkk, 2020-11-02:
You have to click on the light (0x680ae2fffe364af_light)

worker72, 2020-11-02:
OMG. Of course. So it was working all the time 
Thanks

closed 2020-11-02 by worker72

karia001, 2020-11-02:
I have the same device and also followed as worker72, only my device still says not supported in the log.
What can I do to fix it

karia001, 2020-11-03:
nov 03 09:53:10 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:10: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUBEEZ"}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"dateCode":"NULL"}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Received Zigbee message from '0x680ae2fffe3a6056', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.5.3_r47"}' from endpoint 1 with groupID 0
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:debug 2020-11-03 09:53:11: Skipping message, definition is undefined and still interviewing
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:info  2020-11-03 09:53:11: Successfully interviewed '0x680ae2fffe3a6056', device has successfully been paired
nov 03 09:53:11 raspberrypi npm[21138]: Zigbee2MQTT:warn  2020-11-03 09:53:11: Device '0x680ae2fffe3a6056' with Zigbee model 'LUBEEZ-12AB' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

can you please help

worker72, 2020-11-03:
Maybe I can help. Did you find the devices.js file, and added the Lubeez device?

karia001, 2020-11-03:
@worker72 i finally got it working, i did exactly the same as you did, added your exact copy of lines to the devices.js but cause it was not working at first i updated to 116 version but forgot to add it to the devices.js again, after adding it works!
the only thing what is not working is the brightness dimming but that might be that my light does not support dimming,:-(
thank you for your offer to help me out

worker72, 2020-11-03:
oh yes, I went through that as well :(
Glad to hear you have it working 👍

Koenkk, 2020-11-03:
Added support for this device.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)