Koenkk/zigbee2mqtt #4838: Bitron Video Thermostat AV2010/32 doesn't send data

Grillfreund, 2020-11-01:
## What happened
I have 8 AV2010/32 thermostats. 4 of them are working as expected. 4 don't send any data. It is possible to set the new temperature for a room via z2m. I see the results on the devices display. But I don't receive the new value back. I also don't get the current room temperature.

With my old smart home gateway Al of them worked correctly.

## What did you expect to happen
All thermostats should work in the same way.

## How to reproduce it (minimal and precise)
Not sure. I already resetter the devices several times an rejoined the network. I have the issue always with the same devices.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: slaesh’s CC2652RB (Sam issue appeared with CC2531
Adapter firmware version: current

Just a conclusion: I really like z2m. It's much more stable than my old "professional" solution. Thank you for this great tool!

closed 2020-12-09 by github-actions[bot]

sunflowerABB, 2020-12-27:
I have the same issue, several of my Bitron AV2010/32 thermostats also no longer send their temperature reading via z2m but retain an old/stale value. Could it be that the thermostats are broken, although I find it unlikely...

Grillfreund, 2020-12-27:
Hi, 
I found a solution for me. I cannot say what's the important step, but with this it worked again:

1) reset all Bitrons and leave the ZigBee network
2) removing the batteries of all Bitrons
3) change network key in z2m
4) reconnect all your other ZigBee devices first
5) connect your Bitrons again